ros2/ros2_documentation #2604: Build Humble on Win10 fails due to file path limit.

genevanmeter, 2022-05-20:
Clean Windows 10 install.  Attempting to follow the documentation verbatim.  
rosidl_typesupport_introspection_tests fails to build due to Windows file path limit. 
Resolved by setting LongFilePaths registry key (https://docs.microsoft.com/en-us/windows/win32/fileio/maximum-file-path-limitation?tabs=cmd).

Suggesting to adding a note/additional step about Windows file path limit to the Windows build documentation.

christophebedard, 2022-05-20:
> Attempting to follow the documentation verbatim.

Just to make sure, this means you're using `--merge-install` with `colcon build`?

genevanmeter, 2022-05-20:
Yes.

To narrow it down this output when LongFilePaths has not been set:
```
C:\dev\ros2_humble>colcon build --merge-install --packages-select rosidl_typesupport_introspection_tests --event-handlers console_direct+
Starting >>> rosidl_typesupport_introspection_tests
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
-- Found ament_cmake: 1.3.2 (C:/dev/ros2_humble/install/share/ament_cmake/cmake)
-- Found ament_lint_auto: 0.12.4 (C:/dev/ros2_humble/install/share/ament_lint_auto/cmake)
-- Found rosidl_adapter: 3.1.3 (C:/dev/ros2_humble/install/share/rosidl_adapter/cmake)
-- Found gtest sources under 'C:/dev/ros2_humble/install/src/gtest_vendor': C++ tests using 'Google Test' will be built
-- Added test 'copyright' to check source files copyright and LICENSE
-- Added test 'cppcheck' to perform static code analysis on C / C++ code
-- Configured cppcheck include dirs: C:/dev/ros2_humble/src/ros2/rosidl/rosidl_typesupport_introspection_tests/include/;C:/dev/ros2_humble/src/ros2/rosidl/rosidl_typesupport_introspection_tests/test
-- Configured cppcheck exclude dirs and/or files:
-- Added test 'cpplint' to check C / C++ code against the Google style
-- Configured cpplint exclude dirs and/or files:
-- Added test 'lint_cmake' to check CMake code style
-- Added test 'uncrustify' to check C / C++ code style
-- Configured uncrustify additional arguments:
-- Added test 'xmllint' to check XML markup files
-- Configuring done
-- Generating done
-- Build files have been written to: C:/dev/ros2_humble/build/rosidl_typesupport_introspection_tests
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  rosidl_typesupport_introspection_tests__rosidl_generator_c.vcxproj -> C:\dev\ros2_humble\build\rosidl_typesupport_introspection_tests\Release\rosidl_typesupport_introspection_tests__rosidl_generator_c.dll
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppBuild.targets(382,5): error MSB3491: Could not write lines to file "rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_c.dir\Release\rosidl_t.62982093.tlog\rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_c.lastbuildstate". Path: rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_c.dir\Release\rosidl_t.62982093.tlog\rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_c.lastbuildstate exceeds the OS max path limit. The fully qualified file name must be less than 260 characters. [C:\dev\ros2_humble\build\rosidl_typesupport_introspection_tests\rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_c.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppBuild.targets(382,5): error MSB3491: Could not write lines to file "rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_cpp.dir\Release\rosidl_t.00D4EF3C.tlog\rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_cpp.lastbuildstate". Path: rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_cpp.dir\Release\rosidl_t.00D4EF3C.tlog\rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_cpp.lastbuildstate exceeds the OS max path limit. The fully qualified file name must be less than 260 characters. [C:\dev\ros2_humble\build\rosidl_typesupport_introspection_tests\rosidl_typesupport_introspection_tests__rosidl_typesupport_introspection_cpp.vcxproj]
Failed   <<< rosidl_typesupport_introspection_tests [10.1s, exited with code 1]

Summary: 0 packages finished [12.1s]
  1 package failed: rosidl_typesupport_introspection_tests
```

clalancette, 2022-10-25:
See #3130, which both shortens the path and provides a link to the instructions on how to increase the path limit.

PR #3130 (2022-10-26): Shorten the Windows build-from-source path.
This is to keep the total length down below the 260 character limit imposed by Windows by default.

This should fix #2604
closed 2022-10-26 by clalancette