ros2/ros2_documentation #2616: Build failed in the section"Expanding on ROS 2 interfaces "

shaopengyuan, 2022-05-21:
After creating the package named `more_interfaces`, I followed the guidance to build the package using
`colcon build --packages-up-to more_interfaces`
But it failed, the error message is as follows:
```
Starting >>> more_interfaces
--- stderr: more_interfaces                         
CMake Error at /opt/ros/galactic/share/rosidl_cmake/cmake/rosidl_target_interfaces.cmake:41 (message):
  rosidl_target_interfaces() the second argument 'more_interfaces' must be a
  valid target name
Call Stack (most recent call first):
  CMakeLists.txt:32 (rosidl_target_interfaces)

---
Failed   <<< more_interfaces [0.41s, exited with code 1]
```
It seems that `more_interfaces` is not a valid target name.
How to fix it?

sumedhreddy90, 2022-05-25:
### Please make sure that you have followed all the instructions in the document.
[Document link](https://docs.ros.org/en/galactic/Tutorials/Single-Package-Define-And-Use-Interface.html)
#### There isn't any issue with "more_interfaces is not a valid target name"
#### Make sure your CMakeLists.txt is something similar to the below and don't forget to source it
 . install/local_setup.bash

```
cmake_minimum_required(VERSION 3.8)
project(more_interfaces)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rosidl_default_generators REQUIRED)

set(msg_files
  "msg/AddressBook.msg"
)

rosidl_generate_interfaces(${PROJECT_NAME}
  ${msg_files}
)

ament_export_dependencies(rosidl_default_runtime)

find_package(rclcpp REQUIRED)

add_executable(publish_address_book
  src/publish_address_book.cpp
)

ament_target_dependencies(publish_address_book
  "rclcpp"
)

install(TARGETS publish_address_book
 DESTINATION lib/${PROJECT_NAME})

rosidl_target_interfaces(publish_address_book
  ${PROJECT_NAME} "rosidl_typesupport_cpp")

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # uncomment the line when a copyright and license is not present in all source files
  #set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # uncomment the line when this package is not in a git repo
  #set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()

```
### Latest Build:
```
colcon build --packages-up-to more_interfaces
```
[image]

clalancette, 2022-06-09:
No response in over 2 weeks, so closing.  If you are still having problems, please feel free to reopen.

closed 2022-06-09 by clalancette