ros2/ros2_documentation #2691: link to the arm64 status page is broken

razr, 2022-06-05:
[arm64 link](http://repo.ros2.org/status_page/ros_humble_ufv8.html) is broken on the [Installation page/Resource](https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html#resources). Correct page is https://repo.ros2.org/status_page/ros_humble_ujv8.html

fujitatomoya, 2022-06-06:
ros2 document link is `arm64 <http://repo.ros2.org/status_page/ros_{DISTRO}_ufv8.html>`, this works for everything but humble.

i believe https://repo.ros2.org/status_page/ros_humble_ujv8.html is the wrong link path?

gavanderhoorn, 2022-06-06:
The `f` would be for `focal`, the `j` is for `jammy`.

Notice how Rolling has both `f` and `j` pages. The `f` one is probably a stale page from when Rolling still targetted Focal.

clalancette, 2022-06-06:
Yes, absolutely.  That should be a `j` there.  See #2692 for a fix (which I will also backport to Humble).

PR #2692 (2022-06-06): Fix the link to the arm64 repositories.

Fixes #2691