ros2/ros2_documentation #2730: Python Service Client tutorial example is problematic

SteveMacenski, 2022-06-14:
The tutorial https://github.com/ros2/ros2_documentation/blob/humble/source/Tutorials/Writing-A-Simple-Py-Service-And-Client.rst shows 

```
    while rclpy.ok():
        rclpy.spin_once(minimal_client)
        if minimal_client.future.done():
            try:
                response = minimal_client.future.result()
            except Exception as e:
                minimal_client.get_logger().info(
                    'Service call failed %r' % (e,))
            else:
                minimal_client.get_logger().info(
                    'Result of add_two_ints: for %d + %d = %d' %
                    (minimal_client.req.a, minimal_client.req.b, response.sum))
            break
```

For how to write a client, that's a weird way to do it considering we have `spin_until_future_complete`. I think this needs to be fixed to be following best practices and what's outlined in the examples repo: https://github.com/ros2/examples/blob/master/rclpy/services/minimal_client/examples_rclpy_minimal_client/client.py

```

    future = cli.call_async(req)
    rclpy.spin_until_future_complete(node, future)

```

Maybe the point is to show an example that doesn't block? But I think that we should not be introducing people to basic services with so much code. I think its a 'later' thing to get into if they want to check for the services response in a non-blocking way. 

If you agree, I'm happy to make the update

fujitatomoya, 2022-06-14:
i am not sure if there is any intention on this, but after consensus, it would be nicer if we use `spin_until_future_complete`.

clalancette, 2022-06-14:
> Maybe the point is to show an example that doesn't block? But I think that we should not be introducing people to basic services with so much code. 

Yes, agreed.  As you say, I think we should make this match what is in the examples repository (which will also make it match the C++ documentation)

> I think its a 'later' thing to get into if they want to check for the services response in a non-blocking way.

Yes, a tutorial about how to call services asynchronously and check them without blocking would be a great thing.  We have some language along those lines in https://docs.ros.org/en/rolling/How-To-Guides/Sync-Vs-Async.html , but we should probably clean that up and promote it to a full "Intermediate" tutorial.

SteveMacenski, 2022-06-14:
PR submitted

clalancette, 2022-06-22:
We've now merged in #2731, so closing this out.