ros2/ros2_documentation #2801: [Tutorial] Using ROS Index

Interpause, 2022-06-28:
I think a tutorial should be written about using <https://index.ros.org/>. Right now, most beginners would miss noticing it on <https://docs.ros.org/en/foxy/index.html>. They might also fail to notice shortcuts to the documentation. Ideally, there could also be a tutorial introducing commonly used packages for beginners, such as `image_transport`, even if tutorials for those packages have not been made yet.

audrow, 2022-07-14:
Hi @Interpause, I agree this would be nice. If you're able to and interested, feel free to open a PR adding this tutorial page and @ me for a review.

clalancette, 2024-05-07:
We actually did this, and instead decided to move it over to index.ros.org.  So I'm going to go ahead and close this one out.