ros2/ros2_documentation #2866: Links on humble and rolling About-Client-Interfaces.html point to foxy docs

ihasdapie, 2022-07-12:
Links on the about client interfaces page on ros2 humble and rolling docs still points to foxy. I also tried to find rolling and humble api documentation for `rclpy` but none were to be found. This may be because it seems like Rdoc jobs for `rclpy` docs are [broken](https://build.ros2.org/view/Rdoc/job/Rdoc__rclpy__ubuntu_jammy_amd64/lastBuild/); although the job doesn't fail the documentation job errors out.

[image]

- https://github.com/ros2/ros2_documentation/blame/rolling/source/Concepts/About-Client-Interfaces.rst#L36
- https://github.com/ros2/ros2_documentation/blame/rolling/source/Concepts/About-Client-Interfaces.rst#L54

fujitatomoya, 2022-07-20:
@clalancette 

http://docs.ros2.org should be referred from https://github.com/ros2/ros2_documentation ?

it seems that,

- https://docs.ros2.org/latest/api/ is not up-to-date.
- https://docs.ros2.org/humble/api/ and https://docs.ros2.org/rolling/api/ do not exist.

clalancette, 2022-08-04:
Yes, we shouldn't be using https://docs.ros2.org, and instead should be using the generated API documentation from http://docs.ros.org/en/humble/p/rclcpp/ , for example.  We should update the links here to reflect that.

PR #4389 (2024-05-07): Remove references to docs.ros2.org from the documentation.
There aren't too many left, but the ones that are there should be either removed or pointed to modern places. This commit does exactly that, and outside of release notes there are no more references to docs.ros2.org after this.

This will close #2866 (though only for Iron, Jazzy, and Rolling).
PR #4391 (2024-05-07): Remove references to docs.ros2.org from the documentation. (backport #4389)
There aren't too many left, but the ones that are there should be either removed or pointed to modern places. This commit does exactly that, and outside of release notes there are no more references to docs.ros2.org after this.

This will close #2866 (though only for Iron, Jazzy, and Rolling).<hr>This is an automatic backport of pull request #4389 done by [Mergify](https://mergify.com).
PR #4392 (2024-05-07): Remove references to docs.ros2.org from the documentation. (backport #4389)
There aren't too many left, but the ones that are there should be either removed or pointed to modern places. This commit does exactly that, and outside of release notes there are no more references to docs.ros2.org after this.

This will close #2866 (though only for Iron, Jazzy, and Rolling).<hr>This is an automatic backport of pull request #4389 done by [Mergify](https://mergify.com).
closed 2024-05-07 by ahcorde