ros2/ros2_documentation #2892: example packages are already built in underlay workspaces

ijnek, 2022-07-23:
Following, [Using colcon to build packages tutorial](https://docs.ros.org/en/foxy/Tutorials/Beginner-Client-Libraries/Colcon-Tutorial.html#using-colcon-to-build-packages), the ``colcon build --symlink-install`` command throws warnings before all ros2/examples is already installed by default in a ROS desktop installation. This can be confusing to a beginner.

```sh
[0.435s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'examples_rclpy_minimal_action_client' is in: /opt/ros/rolling
	'examples_rclcpp_minimal_timer' is in: /opt/ros/rolling
	'examples_rclpy_minimal_subscriber' is in: /opt/ros/rolling
	'examples_rclpy_minimal_client' is in: /opt/ros/rolling
	'examples_rclpy_executors' is in: /opt/ros/rolling
	'examples_rclcpp_multithreaded_executor' is in: /opt/ros/rolling
	'examples_rclpy_minimal_service' is in: /opt/ros/rolling
	'examples_rclcpp_minimal_composition' is in: /opt/ros/rolling
	'examples_rclcpp_minimal_client' is in: /opt/ros/rolling
	'examples_rclcpp_minimal_publisher' is in: /opt/ros/rolling
	'examples_rclcpp_minimal_service' is in: /opt/ros/rolling
	'examples_rclpy_minimal_publisher' is in: /opt/ros/rolling
	'examples_rclpy_minimal_action_server' is in: /opt/ros/rolling
	'examples_rclcpp_minimal_action_server' is in: /opt/ros/rolling
	'examples_rclcpp_minimal_action_client' is in: /opt/ros/rolling
	'examples_rclcpp_minimal_subscriber' is in: /opt/ros/rolling
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding examples_rclcpp_minimal_action_client examples_rclcpp_minimal_action_server examples_rclcpp_minimal_client examples_rclcpp_minimal_composition examples_rclcpp_minimal_publisher examples_rclcpp_minimal_service examples_rclcpp_minimal_subscriber examples_rclcpp_minimal_timer examples_rclcpp_multithreaded_executor examples_rclpy_executors examples_rclpy_minimal_action_client examples_rclpy_minimal_action_server examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_minimal_subscriber

This may be promoted to an error in a future release of colcon-override-check.
```

ijnek, 2022-10-29:
Closing, as this issue has been resolved by changing colcon-override-check from a required to recommended dependency in https://github.com/colcon/colcon-common-extensions/pull/15.

closed 2022-10-29 by ijnek

navdotnetreqs, 2024-08-24:
This should have been fixed by also changing the tutorial, imo. Just ran into this w/ iron.