ros2/ros2_documentation #2903: Proposal - rename 'tf2' links in documentation tree and page titles to 'Transformations'

ooeygui, 2022-07-25:
As part of the TF2 documentation refresh, I'd like to rename the top-level nodes for `tf2` to *Transformations* and wanted to discuss before making a change.
[image]

# Reasoning
Newcomers to ROS are likely not be familiar with the tf2 library. I wanted to explain **what and why** of transformations before introducing the `tf2` library. From a documentation perspective `tf2` describes the library of transformations, so I'm proposing rename the root node to *Transformations* and rename *Introduction to tf2* page to *Introduction to Transformations* in order to reduce cognitive burden for new - or returning - users.

clalancette, 2022-07-25:
That seems generally reasonable, though I also want to make sure users who *are* familiar with `tf2` can do a search and find what they are looking for.  How about a compromise of "Transformations (tf2)" ?

ooeygui, 2022-07-25:
(I'll leave this open for a few more days before doing a PR with just this change for other comments.)