ros2/ros2_documentation #2920: Launch tutorial leads to Error on execution

GeoRCode, 2022-08-07:
I was following the tutorial at https://docs.ros.org/en/foxy/Tutorials/Intermediate/Launch/Using-Substitutions.html and everything was going ok until the "Launching example" line: `ros2 launch launch_tutorial example_main.launch.py`
For some reason, this gives me the Errno 2 on the console:
```
FileNotFoundError: [Errno 2] No such file or directory: '/home/user/dev_ws/install/launch_tutorial/share/launch_tutorial/launch/example_substitutions.launch.py'
```
I tried to follow the error towards the source files but couldn't undestand why this is happening.

I also like to point out that in this tutorial when the prerequisites says: "This tutorial also assumes that you have created a new package of ament_python build type called launch_tutorial", it's not really clear to the user if changing setup.py is actually necessary in order to find all the launch files, and if changing package.xml to define its dependencies is needed as well. It's something that I noticed is quite different from the previous tutorials where each step is clear and well defined, regardless of whether this means repeating information and/or commands.

This is the whole console log (just in case is needed too):
```
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2022-08-06-21-42-32-567336-pc-91925
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=FileNotFoundError(2, 'No such file or directory')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 130, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_description_source.py", line 51, in _get_launch_description
    return get_launch_description_from_python_launch_file(location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 62, in get_launch_description_from_python_launch_file
    launch_file_module = load_python_launch_file_as_module(python_launch_file_path)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/python_launch_file_utilities.py", line 37, in load_python_launch_file_as_module
    loader.exec_module(mod)
  File "<frozen importlib._bootstrap_external>", line 844, in exec_module
  File "<frozen importlib._bootstrap_external>", line 980, in get_code
  File "<frozen importlib._bootstrap_external>", line 1037, in get_data
FileNotFoundError: [Errno 2] No such file or directory: '/home/user/dev_ws/install/launch_tutorial/share/launch_tutorial/launch/example_substitutions.launch.py'
```
Hope this helps :)

christophebedard, 2022-08-07:
I think your package doesn't install its launch files (i.e., it doesn't copy them to the right directory under `install/...`), so that's why the launch file is not found. Adding this to the `data_files` list in your `setup.py` file should fix it:

```py
('share/' + package_name + '/launch', glob.glob('launch/*.launch.*')),
```

See the full example here: https://github.com/ros2/ros2_tracing/blob/25142e8cbc58a6317c63014d461d985593ae5c1a/tracetools_launch/setup.py#L14

> I also like to point out that in this tutorial when the prerequisites says: "This tutorial also assumes that you have created a new package of `ament_python` build type called `launch_tutorial`", [...]

I agree, it's skipping over important details. The "Creating a package" tutorial does not mention launch files, and the "Using substitutions" tutorial doesn't mention needing the above change to the `setup.py` file. At the very least, this should be mentioned at the end of the "2 Substitutions example launch file" section or at the beginning of the "Launching example" section.

The ["Creating a launch file" tutorial](https://docs.ros.org/en/humble/Tutorials/Intermediate/Launch/Creating-Launch-Files.html) does [mention `ros2 launch <package_name> <launch_file_name>`](https://docs.ros.org/en/humble/Tutorials/Intermediate/Launch/Creating-Launch-Files.html#ros2-launch), but then simply links to the "Creating a package" tutorial, which again doesn't mention the important bit about launch files. Not sure where it should go.

GeoRCode, 2022-08-08:
Hmm... weird. I actually deleted and created again the package several times but the same error was found always. Maybe it's an error on the tutorial then? I mean, it's not clear as I already said.

About the extra line in `setup.py`, I can see how that can solve the problem, but adding `(os.path.join("share", package_name), glob("launch/*.launch.py"))` to `data_files`, as ["Launching and monitoring multiple nodes tutorial"](https://docs.ros.org/en/humble/Tutorials/Intermediate/Launch/Launch-system.html) actually [mentions](https://docs.ros.org/en/humble/Tutorials/Intermediate/Launch/Launch-system.html#python-packages), shouldn't work as well? I mean, from what I understand, this should search in the `share/package-name` directory for a `launch/` directory and then for all the `.launch.py` files in it, doesn't it?
In fact, now that I give a closer look to the directories in my system, and using the other packages created in the other tutorials with launch files as examples (like [this one](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Using-Parameters-In-A-Class-Python.html#change-via-a-launch-file) or [this one](https://docs.ros.org/en/humble/Tutorials/Intermediate/Launch/Launch-system.html#python-packages) which uses [this previously created package](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Creating-Your-First-ROS2-Package.html)), neither of them creates dependencies at `package.xml` (at least, related to the launch part) or have a `launch/` directory at `install/package-name/share/package-name/` and worked well. In fact, the only difference with the substitution tutorial is that they creates their own node located at `src/package-name/package-name/`, and thus, have a reference in `entry_points` in `setup.py`.

It's that the problem then? It's necessary to create a node for `ros2 launch` to find the launch files? Maybe the path `share/package-name/launch` it's actually wrong? Why it works for the other tutorials then? I'm not sure and I'm kinda confused now. This just sound pretty weird to me.

GeoRCode, 2022-08-08:
Update: I 'think' I just fix it.
The problem was [here](https://docs.ros.org/en/foxy/Tutorials/Intermediate/Launch/Using-Substitutions.html#parent-launch-file), line 19 of the `example_main.launch.py`. Using that line makes `ros2 launch` go and search for `.launch.py` (in this case, `example_substitutions.launch.py`) files in a directory that doesn't exist! The 'fix' consist in just delete the `'launch',` argument.

> It's that the problem then? It's necessary to create a node for `ros2 launch` to find the launch files? Maybe the path `share/package-name/launch` it's actually wrong? Why it works for the other tutorials then? I'm not sure and I'm kinda confused now. This just sound pretty weird to me.

To answer my previous questions, it is not necessary to create a new node, the path in `setup.py` (even tho I couldn't manually found it) is fine and, therefore, that's why it works for the other tutorials. In the other hand, yes, adding `(os.path.join("share", package_name), glob("launch/*.launch.py"))` to the `data_files` list in `setup.py` works and is needed too in order to make the tutorial fully work.

So, the answer to the 'mystery' is that the ["Using substitutions tutorial"](https://docs.ros.org/en/foxy/Tutorials/Intermediate/Launch/Using-Substitutions.html), apart from not being entirely user friendly as the other tutorials, it's actually wrong. Should I publish a commit or something then? Let me know.
Thanks for the time btw.

krislamont, 2022-09-27:
I ran into the same issue. Under [2 Parent launch file](https://github.com/ros2/ros2_documentation/blob/humble/source/Tutorials/Intermediate/Launch/Using-Substitutions.rst#id14) the tutorial shows this line:

```python
PathJoinSubstitution([
    FindPackageShare('launch_tutorial'),
    'launch',
    'example_substitutions.launch.py'
])
```

Everything worked for me when I removed the `'launch'` line:

```python
PathJoinSubstitution([
    FindPackageShare('launch_tutorial'),
    'example_substitutions.launch.py'
])
```

FWIW, this is also what is reflected in the full `example_main.launch.py`.

clalancette, 2022-09-27:
I see what you mean.  That particular problem is solved in https://github.com/ros2/ros2_documentation/pull/3044

clalancette, 2022-09-27:
> It's questionable how 'the fixes' are located at the last tutorial of the whole [Launch section](https://docs.ros.org/en/foxy/Tutorials/Intermediate/Launch/Launch-Main.html). It doesn't seems right.

So it is not really that putting the launch files in a `launch` subdirectory is a "fix", so much as it is using a different convention.  Both will work, it is just a matter of what the best practice is.  I'll bring this up with the rest of the ROS 2 team and see what we want to do here.

clalancette, 2023-06-20:
This was fixed in #3044, so closing this out.  Feel free to reopen if you are still having a problem.

closed 2023-06-20 by clalancette