ros2/ros2_documentation #2921: extra self.subscription in tf2 tutorial

Usimian, 2022-08-08:
Perhaps I'm missing something but I don't see the reason for the extra "self.subscription" in the tutorial
for a python tf2 broadcaster, see below:

        self.subscription = self.create_subscription(
            Pose,
            f'/{self.turtlename}/pose',
            self.handle_turtle_pose,
            1)
        self.subscription     <<<<<<<<<< Why is this here?

    def handle_turtle_pose(self, msg):

christophebedard, 2022-08-08:
The code is copied from here: https://github.com/ros/geometry_tutorials/blob/37420516b9aa4feb3d6d2f7566643dea2fd9fa7a/turtle_tf2_py/turtle_tf2_py/turtle_tf2_broadcaster.py

Looks like it's a simple mistake from when the tutorial was created: https://github.com/ros/geometry_tutorials/pull/34/files#diff-1cdaa036d32a0780aa1f61086f52ea49448669e9efabd29d66e6c1ab78a09a25

We could remove that line in the tutorial and in the original code.

Usimian, 2022-08-08:
It still runs fine.  I was just wondering if there’s more to it. Cheers.

kurshakuz, 2022-08-09:
> It still runs fine. I was just wondering if there’s more to it. Cheers.

Yeah, there is a simple reason behind this line actually. It is used to suppress the "unused variable" warning from displaying. 

You can find the same line on [Writing a simple publisher and subscriber (Python)](https://docs.ros.org/en/rolling/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.html#id1) tutorial.

kurshakuz, 2022-08-09:
I only edited that part in the documentation [in this PR](https://github.com/ros2/ros2_documentation/pull/2922) as updating the `geometry_tutorials` package will require additional unnecessary work

PR #2922 (2022-08-09): add explanation to the self.subscription call in Writing a broadcaster
Closes #2921
closed 2022-08-09 by kurshakuz