ros2/ros2_documentation #2935: tf2 listener tutorial error importing tf2_ros part - tf2 timeout tutorial is ENTIRELY wrong

GeoRCode, 2022-08-11:
Hi.

I've been doing and reviewing all the ros2 tutorials for a while and now I've came across with the tf2 ones.
In first place, the [Listener tutorial](https://docs.ros.org/en/foxy/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Listener-Py.html), specifically on the ["Writing a listener node"](https://docs.ros.org/en/foxy/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Listener-Py.html#write-the-listener-node) where all the main code relies, I noticed some weird stuff that VSCode later confirmed me:

[image]
As you can see, there is an import to a part of  `tf2_ros` module that doesn't actually exists. Then, there is calls to a variable that is defined later on the code (i.e. first it's called, then defined. Weird). At the end, I'm not sure if all of this makes any difference because the node works as expected and you can do the examples normally.

On second place (and the more absurd one), on the [timeout tutorial](https://docs.ros.org/en/foxy/Tutorials/Intermediate/Tf2/Learning-About-Tf2-And-Time-Py.html) nothing inside this tutorial makes any sense considering it is using the listener tutorial as a reference. It asks to change a code that is not found on the listener one and to modify things that already are missing in the original one.

This is not the first time I discover weird things in the ros2_documentation and I would like to suggest a review on this topic, specially in the tutorials part, because is where the common user goes first to find answers or goes to learn things in order to develop new projects using ROS.

Hope this can be solved.

clalancette, 2022-08-11:
> As you can see, there is an import to a part of `tf2_ros` module that doesn't actually exists. Then, there is calls to a variable that is defined later on the code (i.e. first it's called, then [...]

As far as I can tell, that part of the module exists just fine.  Running locally:

```
$ python3 -c 'from tf2_ros import TransformException'
```

We do some slightly non-standard things in the `tf2_ros` python module, so it may be the case that it is confusing VSCode.  So I'm going to ignore this one, as this works as far as I can tell.

> On second place (and the more absurd one), on the [timeout tutorial](https://docs.ros.org/en/foxy/Tutorials/Intermediate/Tf2/Learning-About-Tf2-And-Time-Py.html) nothing inside this tutorial makes [...]

Yes, agreed.  That tutorial needs a review and a rewrite.  I'm going to update the title of this issue to reflect that.  If you have time to look into it and open a PR, it is much appreciated.

> This is not the first time I discover weird things in the ros2_documentation and I would like to suggest a review on this topic, specially in the tutorials part, because is where the common user goes first to find answers or goes to learn things in order to develop new projects using ROS.

Sorry, this comment isn't very helpful.  If you have specific issues with the tutorials, please feel free to open up issues.  But a general "I found weird stuff" doesn't really help us solve anything.

GeoRCode, 2022-08-11:
Oh, hey @clalancette. Thanks for taking the time to review this.
> Yes, agreed. That tutorial needs a review and a rewrite. I'm going to update the title of this issue to reflect that. If you have time to look into it and open a PR, it is much appreciated.
>
Well, I would love to but I'm not sure how. I mean, I've been doing tests on the `turtle_tf2_listener.py` locally and nothing is working so far. The only relevant thing that I discovered so far is that there's a difference between using the line where variable `now` is defined as `now = rclpy.time.Time()` and `now = self.get_clock().now()` (as [here](https://docs.ros.org/en/foxy/Tutorials/Intermediate/Tf2/Learning-About-Tf2-And-Time-Cpp.html#tf2-and-time) is suggested). The difference that I'm talking about is on the `clock_type` that is sent by the listener itself:
With the first option the output is `clock_type=SYSTEM_TIME` and with the other one is `clock_type=ROS_TIME`.

Also, the duration of the timeout (when included) seems to be taken into consideration only when the `ROS_TIME` is used. In that scenario, it never works no matter how many or how little time you put, resulting in the listener incapable of transforming turtle2 to turtle1 saying "Lookup would require extrapolation into the future".

Not sure how to proceed from there.

> Sorry, this comment isn't very helpful. If you have specific issues with the tutorials, please feel free to open up issues. But a general "I found weird stuff" doesn't really help us solve anything.
>
I'm sorry, that was me trying to conclude the issue with a complementary opinion. I've been doing separate issue threads with the things I've been founding in here though.
But yeah, excuse me again for the commentary.

clalancette, 2022-08-11:
> Also, the duration of the timeout (when included) seems to be taken into consideration only when the `ROS_TIME` is used. In that scenario, it never works no matter how many or how little time you [...]

Yeah, that's fine.  When I reviewed it a couple of months ago, that's where I got to as well.  There should be a way to make it work, I just haven't had time to come back and figure out what the tutorial is actually trying to accomplish.

> I'm sorry, that was me trying to conclude the issue with a complementary opinion. I've been doing separate issue threads with the things I've been founding in here though.
> But yeah, excuse me again for the commentary.

No worries, it happens.  We rely a lot on the community to find and fix issues in the documentation, so please do feel free to keep opening issues as you find them.  Thanks for this report.

GeoRCode, 2022-08-11:
> Yeah, that's fine. When I reviewed it a couple of months ago, that's where I got to as well. There should be a way to make it work, I just haven't had time to come back and figure out what the tutorial is actually trying to accomplish.
> 
As far as I understand, the tutorial is trying to teach how the `timeout` on `lookup_transform` works and why is useful. In this case, making something like a delay in order to give the buffer some time to reach the proper response for the listener, in order to take the broadcasted messages as quick a possible and act. Almost like trying to sync up those two I think.
I also think the same as you. There might be a way to make it work, mostly because in cpp there's no issue with it. We'll figure it out eventually.
Thanks for the time! ^^

clalancette, 2024-05-07:
We "fixed" this by removing the tutorial in https://github.com/ros2/ros2_documentation/pull/4384, so closing.

closed 2024-05-07 by clalancette