ros2/ros2_documentation #2944: Autocompletion in zsh failes

Fildo7525, 2022-08-13:
When I used ros2 Foxy on Ubuntu 20.04 in zsh everything was ok. Now I changed to Ubuntu 22.04 and to Humble and the autocompletion is not working for ros2

mjbogusz, 2022-08-22:
https://github.com/ros2/ros2cli/issues/534

endrelaszlo, 2022-12-24:
> Same issue?: [ros2/ros2cli#534](https://github.com/ros2/ros2cli/issues/534)
> 
> However, for me Foxy/Galactic on 20.04 is spotty at best as well

I had the same issue that the current ticket describes (env: Ubuntu 22.04+Humble) and the solution came from the issue referenced by @mjbogusz. Thanks! 

**Solution/workaround** by @chenjunn-s comment: [ros2cli/issues/534#issuecomment-957516107](https://github.com/ros2/ros2cli/issues/534#issuecomment-957516107)  

> By adding these lines to .zshrc the argcompletion is succesfully enabled.
> ```
> # argcomplete for ros2 & colcon
> eval "$(register-python-argcomplete3 ros2)"
> eval "$(register-python-argcomplete3 colcon)"
> ```

Note: If the registration fails, check whether your python environment has the `argcomplete` package installed.

fujitatomoya, 2023-01-15:
@Fildo7525 @endrelaszlo @mjbogusz either of you guys, would you mind if trying https://github.com/ros2/rosidl/pull/702 to see if the problem can be solved?

Fildo7525, 2023-01-15:
endrelaszlo your [comment](https://github.com/endrelaszlo)](https://github.com/ros2/ros2_documentation/issues/2944#issuecomment-1364596865) worked. The only thing I would add is to put this lines before sourcing ros.

SammyRamone, 2023-04-12:
> endrelaszlo your comment [(https://github.com/endrelaszlo)](https://github.com/ros2/ros2_documentation/issues/2944#issuecomment-1364596865) worked. The only thing I would add is to put this lines before sourcing ros.

It also worked for me (Ubuntu 22.04 / Rolling), but only if I put it **after** sourcing ROS. So if somebody else runs into this issue, maybe try both options 🤷

ahamboeck, 2024-10-26:
This works for jazzy / ubuntu 24.04 with zsh

# argcomplete for ros2 & colcon in zsh
eval "$(register-python-argcomplete ros2)"
eval "$(register-python-argcomplete colcon)"