ros2/ros2_documentation #2947: No module named doc8

RFRIEDM-Trimble, 2022-08-15:
During my contribution of this PR #2882, I was unable to build docs due to an issue with finding doc8. I followed the developer instructions exactly. This can easily be fixed by first referencing the user's python interpreter prior to invoking the call to doc8, so it knows where to find it. I will propose a simple fix.

RFRIEDM-Trimble, 2022-08-17:
For a full log trace, it appears to be due to incompatible versions of doc8. 
```
Rfriedm@rfriedm-us-ll2:~/Development/RFRIEDM-Trimble/ros2_documentation$ python3 -m pip install -r requirements.txt 
Collecting doc8
  Downloading doc8-1.0.0-py3-none-any.whl (25 kB)
Requirement already satisfied: docutils==0.16 in /usr/lib/python3/dist-packages (from -r requirements.txt (line 2)) (0.16)
Requirement already satisfied: pip in /usr/lib/python3/dist-packages (from -r requirements.txt (line 3)) (20.0.2)
Collecting sphinx
  Downloading Sphinx-5.1.1-py3-none-any.whl (3.2 MB)
     |████████████████████████████████| 3.2 MB 2.3 MB/s 
Collecting sphinx-copybutton
  Downloading sphinx_copybutton-0.5.0-py3-none-any.whl (12 kB)
Collecting sphinx-multiversion
  Downloading sphinx_multiversion-0.2.4-py3-none-any.whl (9.6 kB)
Collecting sphinx-rtd-theme
  Downloading sphinx_rtd_theme-1.0.0-py2.py3-none-any.whl (2.8 MB)
     |████████████████████████████████| 2.8 MB 2.2 MB/s 
Collecting sphinx-sitemap
  Downloading sphinx-sitemap-2.2.0.tar.gz (6.1 kB)
Collecting sphinx-tabs
  Downloading sphinx_tabs-3.4.1-py3-none-any.whl (10.0 kB)
Requirement already satisfied: Pygments in /usr/lib/python3/dist-packages (from doc8->-r requirements.txt (line 1)) (2.3.1)
Collecting stevedore
  Downloading stevedore-4.0.0-py3-none-any.whl (49 kB)
     |████████████████████████████████| 49 kB 2.2 MB/s 
Collecting tomli; python_version < "3.11"
  Downloading tomli-2.0.1-py3-none-any.whl (12 kB)
Collecting restructuredtext-lint>=0.7
  Downloading restructuredtext_lint-1.4.0.tar.gz (16 kB)
Requirement already satisfied: requests>=2.5.0 in /usr/lib/python3/dist-packages (from sphinx->-r requirements.txt (line 4)) (2.22.0)
Collecting sphinxcontrib-devhelp
  Downloading sphinxcontrib_devhelp-1.0.2-py2.py3-none-any.whl (84 kB)
     |████████████████████████████████| 84 kB 1.7 MB/s 
Collecting Jinja2>=2.3
  Downloading Jinja2-3.1.2-py3-none-any.whl (133 kB)
     |████████████████████████████████| 133 kB 2.8 MB/s 
Collecting sphinxcontrib-jsmath
  Downloading sphinxcontrib_jsmath-1.0.1-py2.py3-none-any.whl (5.1 kB)
Collecting sphinxcontrib-serializinghtml>=1.1.5
  Downloading sphinxcontrib_serializinghtml-1.1.5-py2.py3-none-any.whl (94 kB)
     |████████████████████████████████| 94 kB 1.9 MB/s 
Collecting importlib-metadata>=4.4; python_version < "3.10"
  Downloading importlib_metadata-4.12.0-py3-none-any.whl (21 kB)
Collecting snowballstemmer>=1.1
  Downloading snowballstemmer-2.2.0-py2.py3-none-any.whl (93 kB)
     |████████████████████████████████| 93 kB 837 kB/s 
Collecting babel>=1.3
  Downloading Babel-2.10.3-py3-none-any.whl (9.5 MB)
     |████████████████████████████████| 9.5 MB 1.7 MB/s 
Collecting sphinxcontrib-qthelp
  Downloading sphinxcontrib_qthelp-1.0.3-py2.py3-none-any.whl (90 kB)
     |████████████████████████████████| 90 kB 2.2 MB/s 
Collecting alabaster<0.8,>=0.7
  Downloading alabaster-0.7.12-py2.py3-none-any.whl (14 kB)
Collecting imagesize
  Downloading imagesize-1.4.1-py2.py3-none-any.whl (8.8 kB)
Collecting sphinxcontrib-htmlhelp>=2.0.0
  Downloading sphinxcontrib_htmlhelp-2.0.0-py2.py3-none-any.whl (100 kB)
     |████████████████████████████████| 100 kB 2.1 MB/s 
Collecting sphinxcontrib-applehelp
  Downloading sphinxcontrib_applehelp-1.0.2-py2.py3-none-any.whl (121 kB)
     |████████████████████████████████| 121 kB 2.1 MB/s 
Requirement already satisfied: packaging in /usr/lib/python3/dist-packages (from sphinx->-r requirements.txt (line 4)) (20.3)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from sphinx-sitemap->-r requirements.txt (line 8)) (1.14.0)
Collecting pbr!=2.1.0,>=2.0.0
  Downloading pbr-5.10.0-py2.py3-none-any.whl (112 kB)
     |████████████████████████████████| 112 kB 2.1 MB/s 
Collecting MarkupSafe>=2.0
  Downloading MarkupSafe-2.1.1-cp38-cp38-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (25 kB)
Collecting zipp>=0.5
[... 127 lines cut ...]
Preparing to unpack .../9-libgts-bin_0.7.6+darcs121130-4_amd64.deb ...
Unpacking libgts-bin (0.7.6+darcs121130-4) ...
Setting up liblab-gamut1:amd64 (2.42.2-3build2) ...
Setting up libgts-0.7-5:amd64 (0.7.6+darcs121130-4) ...
Setting up libpathplan4:amd64 (2.42.2-3build2) ...
Setting up libann0 (1.1.2+doc-7build1) ...
Setting up libcdt5:amd64 (2.42.2-3build2) ...
Setting up libcgraph6:amd64 (2.42.2-3build2) ...
Setting up libgts-bin (0.7.6+darcs121130-4) ...
Setting up libgvc6 (2.42.2-3build2) ...
Setting up libgvpr2:amd64 (2.42.2-3build2) ...
Setting up graphviz (2.42.2-3build2) ...
Processing triggers for man-db (2.9.1-1) ...
Processing triggers for libc-bin (2.31-0ubuntu9.9) ...
Rfriedm@rfriedm-us-ll2:~/Development/RFRIEDM-Trimble/ros2_documentation$ make test
doc8 --ignore D001 --ignore-path build
make: doc8: Command not found
make: *** [Makefile:22: test] Error 127
```

Specifically, this excerpt: 
```
ERROR: doc8 1.0.0 has requirement docutils<0.21,>=0.19, but you'll have docutils 0.16 which is incompatible.
ERROR: sphinx-tabs 3.4.1 has requirement docutils~=0.18.0, but you'll have docutils 0.16 which is incompatible.
```

RFRIEDM-Trimble, 2022-08-17:
The following is a fix. Yes the version is not compatible, but the real issue was it not being on path. To fix, I added it to my bash_profile (which is source in ~/.bashrc)
```
Rfriedm@rfriedm-us-ll2:~/Development/RFRIEDM-Trimble/ros2_documentation$ tail -n 1 ~/.bash_profile
export PATH=$PATH:~/.local/bin
```

clalancette, 2022-09-01:
Given that we closed the connected pull request, I'm going to go ahead and close this as well.  If you think this is still a problem, please feel free to reopen.

closed 2022-09-01 by clalancette