ros2/ros2_documentation #2975: Logging throttle

mshahbionaut, 2022-08-20:
## Feature request

#### Feature description
I want to log some info statements using `get_logger().info('your code is working as expected')`, however, logging these in every iteration is causing my node to slow down.
ROS1 had a way to reduce the frequency with which logs were printed using `rospy.log_throttle.info()` IIRC.
However, I have not been able to find this in rclpy for ROS2. It does seem to exist in rclcpp according to [this post](https://answers.ros.org/question/352364/ros2-throttle-logs/).
Would it be possible to implement something similar in ROS2 also?

#### Implementation considerations
I'm not a low level programmer, just an avid user of these awesome tools. Some basic reading of the ROS2 design pages, my understanding is that implementing identical features in rclpy and rclcpp is much easier than implementing identical features in rospy and roscpp, since rclpy and rclcpp are both built on the same C-client, where as rospy and roscpp are completely independent implementations.

Based on this I would assume implementing something in rclpy which already exists in rclcpp should be feasible. Please let me know if this is not true. Also, if someone gave me some guidance, I could try to contribute and submit pull requests myself.

Thanks a lot!

mshahbionaut, 2022-08-21:
@clalancette Thanks a lot! I have been struggling with finding the right documentation for a lot of `rclpy` and `launch` usage. I'm assuming this is because it's still relatively new or not as widely adopted yet.

I'll try to be more diligent about recording issues when I see missing documentation.
Also, if you have any specific guidance beyond the Sphinx page linked in readme on contributing to documentation, happy to help out.

Thanks!

clalancette, 2022-09-28:
This was actually solved via https://github.com/ros2/ros2_documentation/pull/3025 , so closing this out.

closed 2022-09-28 by clalancette