ros2/ros2_documentation #3015: Unused Variable in TF2 Static Broadcaster Tutorial (Confusing)

joshnewans, 2022-09-14:
Currently, the tutorial [here](https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/Intermediate/Tf2/Writing-A-Tf2-Static-Broadcaster-Py.rst) contains the following section:

``` python
   def __init__(self, transformation):
      super().__init__('static_turtle_tf2_broadcaster')

      self._tf_publisher = StaticTransformBroadcaster(self)

      # Publish static transforms once at startup
      self.make_transforms(transformation)

   def make_transforms(self, transformation):
      static_transformStamped = TransformStamped()
      static_transformStamped.header.stamp = self.get_clock().now().to_msg()
      static_transformStamped.header.frame_id = 'world'
      static_transformStamped.child_frame_id = sys.argv[1]
      static_transformStamped.transform.translation.x = float(sys.argv[2])
      static_transformStamped.transform.translation.y = float(sys.argv[3])
      static_transformStamped.transform.translation.z = float(sys.argv[4])
      quat = quaternion_from_euler(
            float(sys.argv[5]), float(sys.argv[6]), float(sys.argv[7]))
      static_transformStamped.transform.rotation.x = quat[0]
      static_transformStamped.transform.rotation.y = quat[1]
      static_transformStamped.transform.rotation.z = quat[2]
      static_transformStamped.transform.rotation.w = quat[3]

      self._tf_publisher.sendTransform(static_transformStamped)

def main():
...
   node = StaticFramePublisher(sys.argv)
...
```

The parameter `transformation` is not used by `make_transforms` and it seems the intention is that it should have been used throughout the function everywhere that `sys.argv` was used (as the C++ version does). The explanation for the code further down in the tutorial doesn't really help to clarify it.

I'm happy to have a go making a PR with this change if people are ok with that (obviously the change is simple, I'm just not familiar with the contribution process, backporting etc beyond what is said [here](https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Contributing-To-ROS-2-Documentation.html)).

---

As an aside, I also ran into issues with the `quaternion_from_euler` function which is used undefined in this tutorial. I discovered #2467 and #2977 which purport to deal with this, but as far as I can tell that PR simply removes the `import` call in the tutorials, leaving an undefined function call. The [Quaternion Fundamentals tutorial](https://github.com/ros2/ros2_documentation/blob/rolling/source/Tutorials/Intermediate/Tf2/Quaternion-Fundamentals.rst) makes a reference to [this other file](https://github.com/ros/geometry_tutorials/blob/ros2/turtle_tf2_py/turtle_tf2_py/turtle_tf2_broadcaster.py) (arguably somewhat truncated), however that tutorial comes after this one (and other similar ones) in the list, and also no mention is made within this tutorial - if you find it via a search and jump right in you just have code that doesn't run.

I assume the people involved in that PR had a plan and something was just left out (or I've misunderstood something) so I don't want to go ahead and propose an alternative, but I'm happy to expand this into its own issue if others agree that it is one.

[image]

clalancette, 2022-09-14:
Thanks for the report.  I'm going to address your concerns in reverse order.

For the `quaternion_from_euler` problem, you are absolutely right that they are missing.  That was just an oversight, and should be fixed by #3020.

For the `make_transforms` problem, that is actually a small bug in the code from https://github.com/ros/geometry_tutorials/blob/ros2/turtle_tf2_py/turtle_tf2_py/static_turtle_tf2_broadcaster.py .  In particular, `make_transforms` should be referencing the `transformation` variable, not using `sys.argv` directly.  I'll take a look at fixing that in the upstream code (and then pulling it into the documentation).

clalancette, 2022-09-14:
And https://github.com/ros/geometry_tutorials/pull/71 should fix the latter problems.

joshnewans, 2022-09-14:
Thanks for the quick response Chris!
I was a bit confused as to the relationship between `ros/geometry_tutorials` and `ros2/documentation/...`, but am I correct in understanding that the former is example code, while the latter is an actual tutorial on how to write that example code?

Either way, both of those fixes look great. I'd just want to add a reminder that when you pull the code into the documentation to not just fix the bit up the top but also later on where it steps through (but I'm sure you're all over that :) ).

Thanks again for the quick fix, I'm happy for you to close this issue if you're happy with it (given that one is now merged and you have an open PR for the other).

clalancette, 2022-09-19:
All right, with https://github.com/ros2/ros2_documentation/pull/3024 now merged and deployed, and the changes in https://github.com/ros/geometry_tutorials/pull/71 , I consider this one done.  The "time" portions of the tf2 tutorials still need some work, though that is tracked separately in #2935 .

Please take a look at all of the updates.  If you find something else that doesn't make sense, don't hesitate to open up a new issue.

closed 2022-09-19 by clalancette

joshnewans, 2022-09-19:
Wow, nicely done! I had a bit of a read through and those improvements look great. Thanks!