ros2/ros2_documentation #3039: How to install `rosdep`?

eleehiga, 2022-09-25:
I have installed and sourced the bash file for ROS2 but when I run the command `rosdep` it yields `-bash: rosdep: command not found`. Therefore, how would you guys recommend me to install rosdep?

clalancette, 2022-09-25:
> I have installed and sourced the bash file for ROS2 but when I run the command `rosdep` it yields `-bash: rosdep: command not found`. Therefore, how would you guys recommend me to install rosdep?

You should do `sudo apt-get install python3-rosdep` to get it.

Note that we are currently working on https://github.com/ros-infrastructure/rep/pull/355, which should make it so that it automatically gets installed when installing `ros-rolling-desktop` in the future.

clalancette, 2023-06-20:
And this actually got done, so I'm closing this out.  If you are still having problems, please open up a new issue.

closed 2023-06-20 by clalancette