ros2/ros2_documentation #3124: Foxy Ubuntu Debian installation should mention how to set up autocompletion

Toefinder, 2022-10-24:
I have recently installed Foxy on my system, following: https://docs.ros.org/en/foxy/Installation/Ubuntu-Install-Debians.html

To my surprise, there is no auto-completion functionality. I only found out about having to install `argcomplete` after looking at the equivalent page for Crystal: https://docs.ros.org/en/crystal/Installation/Linux-Install-Debians.html

I realised `argcomplete` was also mentioned, but hidden in the "Ubuntu Install from source" tab in Foxy documentation
https://docs.ros.org/en/foxy/Installation/Alternatives/Ubuntu-Development-Setup.html

Should we make auto-completion automatic for installing from Debian package? Or perhaps another way to make clear how to configure auto-completion would be much appreciated.

clalancette, 2022-10-24:
Yes, you are absolutely right.  That's an oversight on our part; this was fixed in Galactic and later to be automatically installed, but not in Foxy.  See #3129, which should fix this.

Toefinder, 2022-10-25:
Thank you @clalancette for the quick reply. I will close the issue now since it's been addressed.

PR #3129 (2022-10-25): (Foxy only) Have users install python3-argcomplete.
This is so that tab completion on the command-line works by default for users.

Closes #3124
closed 2022-10-25 by Toefinder