ros2/ros2_documentation #3167: Windows 10 Instructions minor issues - openssl version, visual studio version

emersonknapp, 2022-11-03:
Openssl: Instructions specify `1.1.1n` but the openssl win site always updates versions without leaving old versions available, it's now `1.1.1s`. Maybe we should say "Windows x64 1.1.1<Letter> - NOT LITE" or something like that. It's unfortunately not something we can just link to...

Visual studio: The link that says "Download here" now will only download visual studio 2022, but 2019 is strictly specified by the instructions. You can go to "old version downloads" at https://my.visualstudio.com/Downloads?q=visual%20studio%202019&wt.mc_id=0~msft~vscom~older-downloads - maybe this is where we should point? I'm currently attempting a build (up to rosbag2) with vs2022 and so far so good but we'll see...

I will happily make updates but I'm on the move so this will track these notes for now.

christophebedard, 2022-11-03:
> Visual studio: The link that says "Download here" now will only download visual studio 2022, but 2019 is strictly specified by the instructions. You can go to "old version downloads" at [...]

The link here is already meant to point to VS 2019: https://docs.ros.org/en/rolling/Installation/Windows-Install-Binary.html#install-visual-studio. See #2123. So you're saying that it leads to VS 2022 now?

emersonknapp, 2022-11-04:
> So you're saying that it leads to VS 2022 now?

Yes, once I downloaded via that link only 2022 options were available. Now that I have downloaded the 2019 version specifically, it shows both if I try via the doc'ed link again, but it didn't the first time around.

christophebedard, 2022-11-04:
Oh I see! I did obviously have VS 2019 installed when I tested the link for #2123. We need a proper solution then.

emersonknapp, 2022-11-07:
I'm guessing the "VS Installer" program is pretty much (maybe exactly) the same, but the download has some hardcoded configuration of which menu options / web assets to make available. So once one installer places that config on the system, it's available to the other because they put it in the same place. Just a guess. Regardless, yes we need some long term solution. 

I notice that we `choco install vcredist` for some version in the setup, but i don't think that's all of the VS tooling, just some portion of it? It's been a long time since I did any real windows dev so I'm hazy on the details of how it works. Is there any way that a package manager can install compilers?

sloretz, 2024-08-12:
It looks like it's no longer possible to install Visual Studio 2019 for free

The direct download link in the ROS 2 documentation now redirects to this page: https://visualstudio.microsoft.com/vs/older-downloads/

Which says:

```
Older versions require an active Visual Studio Subscription. Choose a product below and click on the download button to log in to your Visual Studio (MSDN) subscription and gain access to the older versions.
```

I tried signing up for the free "Dev Essentials" subscription, but that's not enough to download Visual Studio 2019 Community. I can see the download listed, but it says it's not available for my subscription. The cheapest paid subscription I see is $1,199 a year.

clalancette, 2024-10-11:
FYI for anyone who gets here; it actually *is* still possible to get MSVC 2019, it is just hidden from their front page.  You can download it from https://aka.ms/vs/16/release/vs_buildtools.exe -OutFile vs_buildtools_2019.exe .

emersonknapp, 2025-04-28:
I think all the windows instructions are changed now and I have no interest in this ticket, so I'll close as probably not relevant.

closed 2025-04-28 by emersonknapp