ros2/ros2_documentation #3168: Tutorial on the use of timers and publisher callbacks

IamPhytan, 2022-11-03:
Hello,

I recently had to migrate a [motor driver node](https://github.com/norlab-ulaval/roboclaw_ros/tree/humble) in ROS2, where the node:
* subscribes to `Twist` messages from a `teleop` node
* publishes a `Odometry` message at a higher frequency than the Twist messages

After searching extensively, I found out that there were not a lot of online examples that had this kind of publish-subscribe with different frequencies, as most examples have subscriber callback functions that publish messages in the body of the function. 

My colleagues and I finally found out that we could use a publish method as a timer callback, and I think it could be great to add a tutorial on the use of timers for that, as it is a problem that is fairly common. Here is [an example node](https://gist.github.com/IamPhytan/c1c8c976399e562fa38d6ab43d83513e) that I made in order to find solutions to our problem.

Thanks

clalancette, 2022-11-04:
https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Writing-A-Simple-Py-Publisher-And-Subscriber.html#write-the-publisher-node has an example of using `publish` in a timer callback.  Is that enough of an example?

fujitatomoya, 2022-11-08:
IMO, I would like to keep `examples` and `demos` as simple as possible to `how to call or demonstrate ROS 2 APIs or Features`, not so much related to how to write the application.