ros2/ros2_documentation #3169: Humble requires C++17 but docs use C++14

PAJohnson, 2022-11-03:
According to this page: https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027 Humble requires C++17. The docs use C++14 in the `CMakeLists.txt` listings.

The examples work, but other code I've written failed to compile due to `std::variant` usage in `rclcpp`. Changing to C++17 fixed this issue.

Is there any reason to keep C++14 as the standard in the Humble docs? If not, I'll fork and open a PR with the standard set to 17 everywhere.

clalancette, 2022-11-04:
> According to this page: https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027 Humble requires C++17. The docs use C++14 in the `CMakeLists.txt` listings.

It's slightly tricky.  The idea is that we *allow* C++17 compatibility, but don't actually require packages to enable that.  If you look at all of the core packages, the vast majority of them are still C++14.  If we need to switch to C++17 (like we are slowly doing in Rolling), then we can and will.

Because of that, all of the examples in the sources are C++14, and that is why the examples in the docs are still C++14.

In short, I'd say that we should continue to have the examples be C++14 (for consistencies sake), but there is no problem at all for you to use C++17 for your own projects.

PAJohnson, 2022-11-04:
Understood, thanks for the quick reply!

closed 2022-11-04 by PAJohnson