ros2/ros2_documentation #3263: Add ros2 service info

fujitatomoya, 2022-12-31:
document follow up for https://github.com/ros2/ros2cli/pull/771.

after https://github.com/ros2/ros2cli/pull/771 is merged, https://docs.ros.org/en/rolling/Tutorials/Beginner-CLI-Tools/Understanding-ROS2-Services/Understanding-ROS2-Services.html#ros2-service-type needs to be updated accordingly.

lonebots, 2023-01-24:
Hi, I would like to help you one this, although I am new to ros2 I think I can help!

fujitatomoya, 2023-01-24:
@lonebots thanks! once https://github.com/ros2/ros2cli/pull/771 is merged, we can get back to this thread to discuss where and what to add for documentation.

SrijaneeBiswas, 2023-02-27:
First time contributor. Willing to help if more than one contributor is needed.

ashishmiryalkar, 2023-10-11:
Hi, I have some experience with ros, I would like to contribute if needed.

fujitatomoya, 2023-10-11:
@SrijaneeBiswas @ashishmiryalkar thank you very much! we are almost there, but https://github.com/ros2/ros2cli/pull/771 has not been completed yet. let me ping you once https://github.com/ros2/ros2cli/pull/771 is merged ❗ thanks

yaswanth1701, 2023-12-16:
Hi @fujitatomoya , I think the PR you mentioned has been merged. If, yes I would like to work on this issue if no one else is working on it.

fujitatomoya, 2023-12-19:
@yaswanth1701 thanks! i do not know if anyone is working on this. i am happy to review 👍

yaswanth1701, 2023-12-19:
thanks a lot, will start working on this.

fujitatomoya, 2023-12-26:
@yaswanth1701 thanks for creating PR, i will take a look at it.

PR #4080 (2023-12-26): Added ros2 service info to documentation
closes #3263 
Updated the existing documentation of ROS2 services by adding ```ros2 service info``` description and example usage case.
closed 2023-12-26 by clalancette