ros2/ros2_documentation #3310: Unclear wording under the section 'testing'  in the Developer-Guide

jsduenass, 2023-02-07:
In the [testing section](https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Developer-Guide.html#testing) appears the following pharse
> In general minimizing external or cross package test dependencies should be avoided to prevent circular dependencies and tightly coupled test packages.

With the current wording conveys that minimizing external packages should be avoided.  How about rewording it to

> In general using external or cross package test dependencies should be minimized and avoided to prevent circular dependencies and tightly coupled test packages.

clalancette, 2023-02-07:
Yeah, I agree that is badly worded.  I'd go with:

> In general external or cross package test dependencies should be minimized to prevent circular dependencies and tightly coupled test packages.

If you'd like to open up a PR with that change, I'd be happy to review it.

jsduenass, 2023-02-07:
On it

PR #3312 (2023-02-08): Fix wording in testing section Developer-Guide.rst
Close #3310
commit 7a0060c4 (2023-02-08): Fix wording in testing section Developer-Guide.rst (#3312)

Close #3310
commit ed23a3bb (2023-02-08): Fix wording in testing section Developer-Guide.rst (#3312) (#3316)

Close #3310

(cherry picked from commit 7a0060c40e489953913fb0d54e8c8be8840472a5)
commit d75304c8 (2023-02-08): Fix wording in testing section Developer-Guide.rst (#3312) (#3315)

Close #3310

(cherry picked from commit 7a0060c40e489953913fb0d54e8c8be8840472a5)
closed 2023-02-08 by clalancette