ros2/ros2_documentation #3320: Update ROS 2 real-time documentation

carlossvg, 2023-02-09:
Hi @clalancette, as part of the RTWG we are planning to update ROS 2 real-time documentation.

The plan, as explained in https://github.com/ros-realtime/community/issues/41#issuecomment-1424128043, is to migrate some of already existing docs from https://ros-realtime.github.io/ to https://docs.ros.org/. Before creating a PR we would like to get some feedback about the best way to add the mentioned changes. Specifically, we would like to know which option would be better:
1. Create one PR to updated all the related documents
2. Split the work to multiple PRs, so it is better digestible for review. 

FYI @JanStaschulat @razr @nightduck

clalancette, 2023-02-09:
> The plan, as explained in [ros-realtime/community#41 (comment)](https://github.com/ros-realtime/community/issues/41#issuecomment-1424128043), is to migrate some of already existing docs from https://ros-realtime.github.io/ to https://docs.ros.org/.

Before we start migrating, what is the overall goal here?  Why do we want to move this into the main ROS 2 documentation?  What about linking to the realtime documentation from the `Related-Projects` section?  Would that work?

JanStaschulat, 2023-02-10:
Hi @clalancette, the motivation for consolidating the real-time documentation is:
- improve structure of real-time documentation. Currently, there are bits and pices all over the place. Some information in Tutorial some in HowTo on docs.ros.org and some in the RTWG space. The goal is to have an entry point in Tuturial/real-time and consolidate documentation and give pointers to RTWG pages there, where suitable.
- Some information is missing, especially for beginners, which features are available in ROS 2 to get the most out of it for their real-time application. 
- The reason to move some information on ros-realtime.github.io to docs.ros.org is better visibility. If someone searches for real-time, there is hardly any information on docs.ros.org. 
- We want to realize to call of the ROS 2 TSC to improve the ROS 2 documentation. I think, that it is better to have well structured documentation on docs.ros.org rather than on some community repository only. 
- real-time documentation could certainly be added to `Related-Project`, but this alone won't fix the problems described above (in my opinion)

JanStaschulat, 2023-02-13:
Specifically, we propose to:
- move one tutorial about "ROS 2 Tracing" from ros-realtime to mainline
- create a subgroup “Real-Time” under “Tutorial/Advanced” with two new Tutorials about Real-Time Development
- add one page under Concepts with some Background information about Real-Time in ROS 2.

See [here](https://docs.google.com/document/d/1FLv5ggb9qODqGIzNT1wHOOR7JrENRBJsFN8XQq78crM/edit?usp=sharing) for more details.

fujitatomoya, 2023-02-13:
@JanStaschulat @carlossvg @clalancette 

Basically I will second this, having `Real-Time` related documentation for end users will be useful. and entry point should be linked to here to guide the `Real-TIme` users from central documentation site.

Here are my questions and opinions,

- if the `Tutorial` is not requiring special system configuration such as RT patch (only having vanilla ubuntu and install ROS 2 binary packages), that said general `Real-Time` tutorial with ROS 2 that enhances `Real-Time` capability for the application, i think it can be in `Tutorial/Advanced` in here.
- if the `Tutorial` is requiring special system configuration, maybe that should be aligned with `Real-Time` set up procedure in `Real-Time` documentation site. I would think for those users, they would probably have a scope for `Real-Time` system before this `Tutorial`, so it would be better that we have these documentation under `Real-Time` documentation site as well? if i were reading this doc, this would be probably more `user-friendly` structure to avoid jumping back and forth documentation sites.
- `ROS 2 Tracing` can be in main doc site, off topic from `Real-Time`? I believe `Tracing` capability and `Real-Time` is not really connected directly. and besides, these tracing points are already integrated in the core libraries.

> What about linking to the realtime documentation from the Related-Projects section? Would that work?

at least this makes sense to me, not to scatter the ROS 2 documentation.

christophebedard, 2023-02-13:
> * `ROS 2 Tracing` can be in main doc site, off topic from `Real-Time`? I believe `Tracing` capability and `Real-Time` is not really connected directly. and besides, these tracing points are already integrated in the core libraries.

Yeah, I can take care of moving the existing tutorial from ros-realtime.github.io ([here](https://ros-realtime.github.io/Guides/ros2_tracing_trace_and_analyze.html)) to docs.ros.org separately.

clalancette, 2023-02-13:
> * if the `Tutorial` is not requiring special system configuration such as RT patch (only having vanilla ubuntu and install ROS 2 binary packages), that said general `Real-Time` tutorial with ROS 2 [...]

That seems reasonable.  That said, there is also https://github.com/ros2/realtime_support/issues/123 , where we are trying to spin realtime support *out* of the core, so I want to be consistent.

> * if the `Tutorial` is requiring special system configuration, maybe that should be aligned with `Real-Time` set up procedure in `Real-Time` documentation site. I would think for those users, they [...]

Yeah, I think for these things we want to put them on the ros-realtime site.

> * `ROS 2 Tracing` can be in main doc site, off topic from `Real-Time`? I believe `Tracing` capability and `Real-Time` is not really connected directly. and besides, these tracing points are already integrated in the core libraries.

Yeah, agreed, and it looks like Christophe is already on this.