ros2/ros2_documentation #3340: Simulation tutorial still advises ignition commands

knmcguire, 2023-02-17:
Hi!

The simulation tutorial still indicates to use the ignition commands which doesn't work after fortress

For instance, if you would follow this for Humble:
- https://docs.ros.org/en/humble/Tutorials/Advanced/Simulators/Gazebo.html

Then it sends to the install instructions of gazebo, which according to [this table](https://github.com/gazebosim/ros_gz) is either Fortress or Humble. If  Fortress is installed, that perhaps `ign gazebo` would still work, but I installed Garden since binaries were available, and there you use `gz sim -v 4 -r visualize_lidar.sdf ` (I still get errors with this but that is perhaps another issue).

I also see that using `ign gazebo` is also presenting in the [Rolling docs](https://docs.ros.org/en/rolling/Tutorials/Advanced/Simulators/Gazebo.html#)

I understand that the whole transition from ignition to gazebo naming is difficult to deal, especially if Fortress is still being in LTS... but would it be an idea to at least change the documentation to use Garden from Humble and Rolling on and indicate that particular that in the tutorial?

fujitatomoya, 2023-02-17:
@knmcguire thanks for posting issue.

i think that makes sense and should be addressed to replace `igz` command with `gz`.
probably we can add separate tab to either `Fortress` or `Garden`?

CC: @clalancette

clalancette, 2023-02-17:
I'd like to get the opinion of the Gazebo team before we do anything here, as the support situation with Gazebo and ROS is currently a little complicated.

@azeey @j-rivero Could you maybe take a look here and advise us on what we should do?

ahcorde, 2023-02-20:
I created this PR https://github.com/ros2/ros2_documentation/pull/3342

azeey, 2023-02-22:
The issue is a little bit tricky. If users install Gazebo Fortress, as is recommended for Humble, they would not be able to use the `gz` tool, only `ign` would work. And we can't make `gz` work because the binary conflicts with Gazebo-classic. On the flip side, if they install Gazebo Garden[^1], the `gz` tool works, but in my local testing `ign` doesn't. This might be a regression, which we can hopefully fix soon, but regardless, simply replacing all `ign` with `gz` in the documentation won't work in all cases, especially in the recommended Humble/Fortress pairing. 

We have just updated the docs on https://gazebosim.org/docs/latest/ros_installation that gives the recommended installation option together with a more detailed explanation. It might be best to link to that and leave the `ign` in `ros2_documentation` until we have Iron/Harmonic.

[^1]: Using Garden with ROS2 required building `ros_gz` from source and would, IMO, be considered an advanced use case, so I don't think that should be the primary focus for documentation.

clalancette, 2023-02-22:
> The issue is a little bit tricky. If users install Gazebo Fortress, as is recommended for Humble, they would not be able to use the `gz` tool, only `ign` would work. And we can't make `gz` work because the binary conflicts with Gazebo-classic. On the flip side, if they install Gazebo Garden[1](#user-content-fn-1-5367f880bf88fd5e67b19acda6dc9760), the `gz` tool works, but in my local testing `ign` doesn't. This might be a regression, which we can hopefully fix soon, but regardless, simply replacing all `ign` with `gz` in the documentation won't work in all cases, especially in the recommended Humble/Fortress pairing.

Thanks for the background, @azeey .

So with that said, I'm going to actually leave this open but on the backlog for now.  I'm going to close @ahcorde 's PR, since that isn't the recommended thing to do at this time.  Then when `ign` eventually goes away, we can move on this one.

If anyone disagrees with this, feel free to reopen the PR that I'm going to close, and/or keep commenting here.

knmcguire, 2023-02-22:
Is it maybe an idea to advise people to install Humble & Garden as standard? Or is that difficult with since it is not LTS?

Perhaps just a note at the tutorial could also be enough just to make people aware, if they installed garden (since that has the binary option )

azeey, 2023-02-22:
> Is it maybe an idea to advise people to install Humble & Garden as standard? Or is that difficult with since it is not LTS?

It is difficult because we don't provide the `ros_gz` binary package for the Humble/Garden pair. Users have to build it from source even though it's critical for using ROS with Gazebo. And we can't provide binaries for `ros_gz` because Garden binaries are not included in packages.ros.org or Ubuntu upstream for the platforms supported by Humble.
 
> Perhaps just a note at the tutorial could also be enough just to make people aware, if they installed garden (since that has the binary option )

I think adding a link to the https://gazebosim.org/docs/garden/ros_installation would be very helpful.

knmcguire, 2023-02-23:
Ah oke, but I think it was possible to install the garden binaries with sudo apt-get install ros-humble-gz-ros ? or is that only supposed to work with Fortress?

Sorry if I misunderstand anything here

azeey, 2023-02-23:
hmm, I just checked to be sure, but when I try to install `ros-humble-ros-gz`, it shows it'll install Fortress as a dependency (eg. libignition-gazebo6), not Garden (eg. libgz-sim7). Can you verify?

knmcguire, 2024-04-19:
ah wow... sorry this completely fell through the cracks. I just noticed I never answered this... 

I just got into setting up a new system for ROS2 and gazebo, and I tried it again. Yes you are right libignition-gazebo6 is the dependency on ROS2 humble.

What is confusing is that [this installation page on gazebo sim ](https://gazebosim.org/docs/harmonic/getstarted)advises harmonic for 22.04 (perhaps at the time that was Fortress?), however, the [readme of ros_gz advises](https://github.com/gazebosim/ros_gz) fortress.

I've heard of more people falling into this trap. How about that we advise people to install gazebo via `sudo apt-get install ros-humble-gz-ros` only in the tutorial?

clalancette, 2024-04-19:
So I believe that this is all going to be different in Rolling/Jazzy.  In particular, we are vendoring gazebo into ROS now, so it should all be `gz` commands then, and it should be a consistent version (Harmonic).  @azeey Can you confirm?

As for Humble and Iron, I'll leave that to @azeey to figure out what the ROS 2 documentation should say about integration with Gazebo.

azeey, 2024-04-19:
Yes, it will be more consistent in Harmonic/Jazzy and we'll only be using `gz` commands. But the issue about which version of Gazebo to run with ROS might come up again in the future. @knmcguire the main installation page advises Harmonic because that's the latest version of Gazebo available. And that's the recommendation if you use Gazebo by itself (without ROS). If you want to use it with ROS, there's a specific pairing of Gazebo versions with ROS versions. For ROS 2 Humble for example, it's Fortress. I think this is described well in https://gazebosim.org/docs/harmonic/ros_installation#picking-the-quot-correct-quot-versions-of-ros-amp-gazebo. It's possible to use other pairings as well, but it requires more work (e.g. building `ros_gz` from source)

Regarding the documentation in this repo, like you said, maybe we can just say install `ros-humble-gz-ros` and it'll install the correct version of Gazebo for you. And as a footnote, we can say you can use newer versions of Gazebo if you really want, but it requires more work and link to the https://gazebosim.org/docs/harmonic/ros_installation page.

knmcguire, 2024-04-22:
I've made a draft PR but there are some  build issues that it hits currently. However, I just wanted to check if this is on the right track?

clalancette, 2024-05-06:
Closing this since #4311 fixed it for Rolling and the Jazzy backport is just about to be merged.

closed 2024-05-06 by clalancette