ros2/ros2_documentation #3369: Launch file different formats : Including XML and YAML files in a launch file

IamPhytan, 2023-03-06:
Hello, 

First off, it's a great idea to have a page to [compare Python, XML and YAML](https://docs.ros.org/en/humble/How-To-Guides/Launch-file-different-formats.html) implementations of a same launch file. Yet, it seems that these implementations doesn't include examples on how to include XML and YAML launch files.

Would it be a good idea to add such examples to the implementations ?

Thanks

fujitatomoya, 2023-03-07:
I think there are already some examples for those? https://github.com/ros2/ros2_documentation/blob/rolling/source/How-To-Guides/Launch-file-different-formats.rst#launch-file-examples includes `Include another launch file` for Python, XML and YAML?

IamPhytan, 2023-03-07:
It seems that `Launch file examples` doesn't show examples on how to include XML and YAML launch files in a Python launch file

fujitatomoya, 2023-03-07:
> examples on how to include XML and YAML launch files in a Python launch file

Ah I see, 👍

clalancette, 2023-03-07:
Yeah, I think it would be fine to show it including XML and YAML files.  If you'd like to contribute a PR to make that change, I'd be happy to review it.

lonebots, 2023-03-08:
As a first time contributor, if more than 1 person is required to do this task, I am happy to help!

IamPhytan, 2023-03-10:
This issue has been fixed by f96274c7d4e08939730d2cbff82f04ca478e86e4.
I'm closing it.

closed 2023-03-10 by IamPhytan

fujitatomoya, 2023-03-11:
@IamPhytan thanks for the contribution 👍