ros2/ros2_documentation #3385: ros2 and webots did not connect with each in tutotial "Setting up a robot simulation (Webots)"

zilong-ding, 2023-03-15:
[image]
I follow the tutotial step by step,and launch the ros2 launch file,and then pub the topic as below 
[image]
but the webots has no response,I donot know why?
because the examples on webots_ros2 run correctly

zilong-ding, 2023-03-15:
the simulation time is always zero,is it this reason?

clalancette, 2023-03-23:
@omichel Can you take a look?

omichel, 2023-03-24:
I will let the @ygoumaz answer here.

ygoumaz, 2023-03-24:
This issue was already addressed on the `webots_ros2` repository. You can check out [the explanations](https://github.com/cyberbotics/webots_ros2/issues/675) for more information.

closed 2023-03-24 by zilong-ding