ros2/ros2_documentation #3388: Can't build on readthedocs.org

nainglinwai1, 2023-03-16:
I wanna test offline format on readthedocs.org but I can't build on the website.

nainglinwai1, 2023-03-17:
`Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/ros2-documentation43/envs/latest/lib/python3.7/site-packages/sphinx/cmd/build.py", line 276, in build_main
    app.build(args.force_all, filenames)
  File "/home/docs/checkouts/readthedocs.org/user_builds/ros2-documentation43/envs/latest/lib/python3.7/site-packages/sphinx/application.py", line 330, in build
    self.builder.build_update()
  File "/home/docs/checkouts/readthedocs.org/user_builds/ros2-documentation43/envs/latest/lib/python3.7/site-packages/sphinx/builders/__init__.py", line 288, in build_update
    len(to_build))
  File "/home/docs/checkouts/readthedocs.org/user_builds/ros2-documentation43/envs/latest/lib/python3.7/site-packages/sphinx/builders/__init__.py", line 300, in build
    updated_docnames = set(self.read())
  File "/home/docs/checkouts/readthedocs.org/user_builds/ros2-documentation43/envs/latest/lib/python3.7/site-packages/sphinx/builders/__init__.py", line 411, in read
    self.env.doc2path(self.config.root_doc))
sphinx.errors.SphinxError: root file /home/docs/checkouts/readthedocs.org/user_builds/ros2-documentation43/checkouts/latest/index.rst not found

Sphinx error:
root file /home/docs/checkouts/readthedocs.org/user_builds/ros2-documentation43/checkouts/latest/index.rst not found`

clalancette, 2023-03-17:
> So, I try to put the fork repo on readthedocs.org for testing the offline download but the build failed during the setup.

Sorry, I still don't know what this means.  Can you please list the exact commands you are running?

nainglinwai1, 2023-03-17:
python -m sphinx -T -E -b html -d _build/doctrees -D language=en . $READTHEDOCS_OUTPUT/html 

here is the last command

tfoote, 2023-04-06:
To reproduce the build yourself please use:

Recommended in a virtualenv: 
```
pip install --no-warn-script-location --upgrade -r requirements.txt
make html
```

Or you can do `make multiversion` to get the site with multiple versions side by side. 

We don't target hosting on readthedocs.org if you want to run it on their infrastructure you'll need to understand what their process is doing differently. We'd be happy to take patches to support that. But as that's not a use case that we use we aren't going to investigate and debug the issues that you may encounter there.

closed 2023-04-06 by tfoote