ros2/ros2_documentation #3405: Add a Dark Theme for the docs pages

Aytoris, 2023-03-25:
Hello, a Dark Mode version of ROS 2 documentations could be a nice addition for many people I think.

clalancette, 2023-03-26:
That would be nice to have.  We won't have time to work on this, but if you can figure out how to do it, please feel free to open a pull request and we'll be happy to review.

Aytoris, 2023-03-30:
I think the theme [sphinx-rtd-dark-mode](https://pypi.org/project/sphinx-rtd-dark-mode/) could be a good fit:
#### config.py
``` Python
extensions = [
    ...
    'sphinx_rtd_dark_mode',
    ...
]
```

However i see some incorrect display whenever a <style> tag is being hard-coded. Examples: Releases.rst ; Governance.rst ; Roadmap.rst

Maybe someone with more experience in CSS can help here.