ros2/ros2_documentation #3445: How can i change the maximum allowed number of parameters for a node?

felipebapa, 2023-04-08:
Hi!
I am new in ROS and ROS2. Now, i am using a driver from Universal Robots in order to move an UR5 to different joint configurations. I have the problem that when I have more than 512 parameters in my .YAML file, I have this error:

Exceeded maximum allowed number of parameters for a node (512), at /tmp/binarydeb/ros-foxy-rcl-yaml-param-parser-1.1.14/src/parse.c:514, at /tmp/binarydeb/ros-foxy-rcl-1.1.14/src/rcl/arguments.c:388

What can I do? Thanks!

clalancette, 2023-04-08:
Since Foxy, we've removed that restriction and you can have unlimited numbers of parameters.  So the easiest thing to do would be to switch to Humble.

If you can't do that, then you'll have to build Foxy from source.  In particular, you'll need to follow the instructions at http://docs.ros.org/en/foxy/Installation/Alternatives/Ubuntu-Development-Setup.html .  And you'll also have to edit https://github.com/ros2/rcl/blob/287ccd9ed06ff5bdded4dfb1130920d592a71bb7/rcl_yaml_param_parser/src/impl/types.h#L34 to increase the value.

In the future, please open questions like this on https://answers.ros.org, which is our central Question and Answer site.  You'll get a better answer there, and it will be searchable for the future.

I'm going to close this one out.

closed 2023-04-08 by clalancette

felipebapa, 2023-04-10:
Thank you so much!