ros2/ros2_documentation #3489: I think "Implementing Custom Interfaces" is missing the .hpp file

DingoOz, 2023-05-01:
The tutorial advised to add this line:
#include "more_interfaces/msg/address_book.hpp"

There is no corresponding header file created and colcon build fails when it can't find that header file.

I suspect the tutorial needs to add in the creation of the .hpp file.

clalancette, 2023-05-01:
> There is no corresponding header file created and colcon build fails when it can't find that header file.

That header file should be automatically generated when you build the message package.  Can you please go back through the tutorial from the beginning and see if you did all steps?  If it is still failing, please leave a detailed log of the steps you took.

clalancette, 2023-06-20:
No response in over a month, so closing.  Please feel free to fix the issues pointed out in review and reopen.  Thanks for the contribution.

closed 2023-06-20 by clalancette