ros2/ros2_documentation #3490: Retrieve LaunchConfiguration as a string

jkaniuka, 2023-05-01:
### Description 
It often happens that we want to use the launch file argument (as a string) as a part of the path to the URDF file. Unfortunately, it is not possible to use the _str()_ method on the _LaunchConfiguration_ object. The problem seems common to me, but the solution is missing from the ROS 2 tutorials page, and can only be found on the ROS Answers forum.

### Use case
Suppose we have a launch file to display the URDF model of the robots from the Star Wars movies - R2D2 and C3P0. It is more convenient to launch the launch file by specifying the robot model (`ros2 launch urdf_pkg display.launch.py robot_model:=r2d2`), rather than the path to the URDF file (`ros2 launch urdf_pkg display.launch.py robot_model:=install/urdf_pkg/share/urdf_pkg/r2d2_model.urdf`).

### Solution
Use **launch.actions.OpaqueFunction** - I've included a short example below. 

```python
import os
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.substitutions import Command
from launch_ros.actions import Node
from launch.substitutions import Command, LaunchConfiguration
from launch.actions import DeclareLaunchArgument, OpaqueFunction

def launch_setup(context, *args, **kwargs):

    robot_model = LaunchConfiguration('robot_model').perform(context) # retrieving the launch configuration, as a string.

    display_model = Node(
        package='robot_state_publisher',
        executable='robot_state_publisher',
        parameters=[{
            'robot_description': Command(['xacro ', os.path.join(get_package_share_directory('urdf_pkg'), robot_model + '_model.urdf')])
        }]
        )

    run_rviz = Node(
        package='rviz2',
        executable='rviz2',
        name='rviz2',
        output='screen',
        arguments=['-d', os.path.join(get_package_share_directory('urdf_pkg'), robot_model + '_config.rviz')]
        )
    
    joint_state_publisher_gui = Node(
        package='joint_state_publisher_gui',
        executable='joint_state_publisher_gui',
        name='joint_state_publisher_gui',
        )

    return [display_model,
            run_rviz,
            joint_state_publisher_gui]

def generate_launch_description():

    return LaunchDescription([
        DeclareLaunchArgument(name='robot_model', default_value='r2d2', choices=['r2d2', 'c3p0']),
        OpaqueFunction(function=launch_setup)
    ])
```

### The problem was mentioned at the following links:

* [ROS2 launch file: how to convert LaunchArgument to string](https://answers.ros.org/question/396345/ros2-launch-file-how-to-convert-launchargument-to-string/)
* [Access launch argument in LaunchFile ROS2](https://answers.ros.org/question/340705/access-launch-argument-in-launchfile-ros2/?answer=340857#post-id-340857)
* [How to access the runtime value of a LaunchConfiguration instance within custom launch code (injected via an OpaqueFunction) in ROS2?](https://answers.ros.org/question/397123/how-to-access-the-runtime-value-of-a-launchconfiguration-instance-within-custom-launch-code-injected-via-an-opaquefunction-in-ros2/)

wjwwood, 2023-05-11:
Well I was a bit wrong, using opaque function technical makes this no prematurely evaluating it but it also makes it impossible to introspect, as anything that happens in the opaque function cannot be looked at by launch. 

Using the configuration as a substitution is still the right answer imo.

jkaniuka, 2023-05-13:
@wjwwood Thanks for the answer :+1: Looking for a way how the launch file argument can be used to make a file path, I came across [**rosetta_launch**](https://github.com/MetroRobots/rosetta_launch) repository by David Lu. He writes there explicitly that 

>  When evaluating the substitutions, lists of strings/substitutions are concatenated. 

In my opinion, people reach for **_OpaqueFunction_** because they want to make a file path assembly using "+" for string concatenation. I would suggest adding the above sentence to the [**Using substitutions**](https://docs.ros.org/en/humble/Tutorials/Intermediate/Launch/Using-Substitutions.html#using-substitutions) tutorial to highlight that you do not need to convert `LaunchConfiguration` to string. 

I removed the OpaqueFunction from my example, and now it seems to look like it should:
```python
    display_model = Node(
        package='robot_state_publisher',
        executable='robot_state_publisher',
        parameters=[{
            'robot_description': Command(['xacro ', PathJoinSubstitution([get_package_share_directory('lab2'), LaunchConfiguration('manipulator_type')]), "_manipulator.urdf" ])
        }]
        )
```

```python
    run_rviz = Node(
        package='rviz2',
        executable='rviz2',
        name='rviz2',
        output='screen',
        arguments=['-d', [PathJoinSubstitution([get_package_share_directory('lab2'), LaunchConfiguration('manipulator_type')]), "_config.rviz"]]
        )
```

closed 2023-05-20 by jkaniuka