ros2/ros2_documentation #3533: Non-inclusive gender options in custom interfaces example

playertr, 2023-05-05:
The [custom interfaces example](https://github.com/ros2/ros2_documentation/blob/humble/source/Tutorials/Beginner-Client-Libraries/Single-Package-Define-And-Use-Interface.rst) in the Beginner Client Libraries tutorial represents gender as a bool, spelled out to be either MALE or FEMALE. ROS users who may have different gender identities may feel excluded by this choice of representation.

```
bool FEMALE=true
bool MALE=false

string first_name
string last_name
bool gender
uint8 age
string address
```

I suggest moving to a named-constant enum with an "other" option or removing the gender field entirely. This change is in keeping with the ROS Community Code of Conduct's preference for [inclusive language](https://osrf.github.io/code_of_conduct/#:~:text=Using%20welcoming%20and%20inclusive%20language.%20We%E2%80%99re%20accepting%20of%20all%20who%20wish%20to%20take%20part%20in%20our%20activities%2C%20fostering%20an%20environment%20where%20anyone%20can%20participate%20and%20everyone%20can%20make%20a%20difference.).

I can make a PR after getting folks' input on this change.

cdhabecker, 2023-05-08:
I concur. Just remove it.
Also, no one stores age -- always store date of birth. 
Also, bool is a poor use case for constants.

Address is complex. How about a basic contact with a phone number? Just a naive, single-number implementation with a type to demonstrate constants.

```
uint8 PHONE_TYPE_HOME=0
uint8 PHONE_TYPE_WORK=1
uint8 PHONE_TYPE_MOBILE=2

string first_name
string last_name
int32 date_of_birth_seconds
string phone_number
uint8 phone_type
```

playertr, 2023-05-08:
Okay, I'll make the PR.

playertr, 2023-05-08:
Okay! I submitted a PR (and I think I followed GPG signing and commit-message conventions -- couldn't find one).

I also deleted the age / date of birth field because IMO the best representation for this field would be a std_msgs/time message, and creating one is outside the scope of this tutorial.

closed 2023-05-09 by clalancette