ros2/ros2_documentation #3587: No Tutorial for Reading from a Bag File in C++

DLu, 2023-05-11:


mroglan, 2023-05-21:
I was looking for a tutorial on this a few weeks ago, and I was eventually able to get a [reader working](https://gist.github.com/mroglan/383cc70a71fbfe7ef93ff6031da8e488). I’d be happy to work on writing a tutorial if nobody is already on it.

clalancette, 2023-05-21:
> I was looking for a tutorial on this a few weeks ago, and I was eventually able to get a [reader working](https://gist.github.com/mroglan/383cc70a71fbfe7ef93ff6031da8e488). I’d be happy to work on writing a tutorial if nobody is already on it.

That would be fantastic!  Thank you.

DLu, 2023-05-24:
@mroglan Thank you for linking your example, its already been helpful to me.

PR #3663 (2023-08-30): Tutorial for Reading from a Bag File in C++
Resolves #3587
closed 2023-08-30 by clalancette