ros2/ros2_documentation #3632: ROS2 software stack on a production robot system

jclinton830, 2023-05-18:
Hi, 

Are there any plans on releasing a tutorial or guide on how to deploy software to a production robot system?

I found this in the Foxy branch - [Packaging your ROS2 application as a snap](https://docs.ros.org/en/foxy/Tutorials/Miscellaneous/Packaging-your-ROS-2-application-as-a-snap.html)

But that is not listed in the humble branch of the documentation. Is it still relevant in Humble?

jkaniuka, 2023-05-20:
Hi @jclinton830 , maybe [**Releasing a Package**](https://docs.ros.org/en/humble/How-To-Guides/Releasing/Releasing-a-Package.html) tutorial will be useful for you (compatible with `humble` distro) :wink:

jclinton830, 2023-05-22:
> Hi @jclinton830 , maybe [**Releasing a Package**](https://docs.ros.org/en/humble/How-To-Guides/Releasing/Releasing-a-Package.html) tutorial will be useful for you (compatible with `humble` distro) wink

I am talking about deploying production software for an application on a robot as an image. What you are referring to is how to release a package to ROS apt package manager to install it directly.

fujitatomoya, 2023-05-22:
@jclinton830 https://docs.ros.org/en/foxy/Tutorials/Miscellaneous/Packaging-your-ROS-2-application-as-a-snap.html is `community-contributed` page, i think nobody has verified that procedure with `snap` and `core22` base. Just guessing that almost same procedure can be applied with `humble` (ubuntu 22.04) but not just confirmed yet.

I am happy to do review, if someone is willing to contribute.

jclinton830, 2023-05-22:
> @jclinton830 https://docs.ros.org/en/foxy/Tutorials/Miscellaneous/Packaging-your-ROS-2-application-as-a-snap.html is `community-contributed` page, i think nobody has verified that procedure with [...]

Fair enough. I will try to follow that tutorial to see if it works in Humble. 

I also found this https://snapcraft.io/docs/ros2-applications issued by Canonical

mjforan, 2023-08-21:
FYI snap is going to be discontinued as of Ubuntu 24.

closed 2023-11-07 by jclinton830