ros2/ros2_documentation #3664: Setup-ROS-2-with-VSCode-and-Docker-Container.rst needs a review

vkuehn, 2023-05-23:
Please review the coumentation Setup-ROS-2-with-VSCode-and-Docker-Container.rst on a clean machine

e.g.
- you name the workspace folder ws_[project] but the follwoing expects /home/ws/src
-Dev Containers: (Re-)build and Reopen in Container was not available for me in VSCode

rlekkerkerker-hva, 2023-12-21:
I do not understand the part  in source/How-To-Guides/Setup-ROS-2-with-VSCode-and-Docker-Container.rst which is quoted below. It says to "create a .devcontainer folder in the root of your workspace", but in the example workspace structure the .devcontainer folder is not in the workspace folder, but it is in the src folder. Which location is correct? I hope @fujitatomoya or @Mario2407 can help me out a bit. Thanks in advance!

`Add a workspace in order to build and open them in a container, e.g.:

cd ~/
mkdir ws_[project]
cd ws_[project]
mkdir src
Now create a .devcontainer folder in the root of your workspace and add a devcontainer.json and Dockerfile to this .devcontainer folder. Additionally, you need to create a cache folder in which you can cache the build and install folders for different ROS 2 distros. The workspace structure should look like this:

ws_[project]
├── cache
|   ├── [ROS2_DISTRO]
|   |   ├── build
|   |   ├── install
|   |   └── log
|   └── ...
|
├── src
    ├── .devcontainer
    │   ├── devcontainer.json
    │   └── Dockerfile
    ├── package1
    └── package2`

fujitatomoya, 2023-12-21:
@vkuehn @rlekkerkerker-hva thanks for posting issue, i will try to take a look later.

bgigous, 2024-02-11:
Adding another part I found confusing:

> Additionally, you need to create a cache folder in which you can cache the build and install folders for different ROS 2 distros.

I do not understand what is meant by "you can cache the build and install folders for different ROS 2 distros." Does that mean I need to copy my ROS 2 distro files to that cache folder? It's not clear.

fujitatomoya, 2024-02-12:
Just a reference, introduced by https://github.com/ros2/ros2_documentation/pull/3343.

> you name the workspace folder ws_[project] but the follwoing expects /home/ws/src

good eye, this is wrong.

this is even not working as expected, it fails since it cannot find the `src` folder via `postCreateCommand`.
i believe that `.devcontainer` needs to be in uner local `workspace` folder.

> Dev Containers: (Re-)build and Reopen in Container

maybe this was available before but currently it should be `Dev Containers: Reopen in Container`.

> you can cache the build and install folders for different ROS 2 distros.

this is not explained well enough, when you open the devcontainer, it will try to bind the local cache folders to the workspace in the container. i guess this is just an option, and user can do that using devcontainer, so i would like to remove them.

> The test misses a startcommand before with interactive container

i think that is expected to open the terminal from vscode, right? so container is already running, i think this should be fine for user?

i will come up with suggested fix for this.

Ryanf55, 2024-02-20:
Does anyone have a simple repository set up which contains a package and the devcontainer according to this guide? I'd like to try it out and see if `gdb` works.

fujitatomoya, 2024-09-23:
I think we can close this with the following PRs merged.

- https://github.com/ros2/ros2_documentation/pull/4161
- https://github.com/ros2/ros2_documentation/pull/4591
- https://github.com/ros2/ros2_documentation/pull/4654

closed 2024-09-23 by fujitatomoya