ros2/ros2_documentation #3733: Remove redirect urls in sitemap.xml

Yadunund, 2023-06-13:
A ticket to discuss ways to remove redirect urls from `sitemap.xml`. Tools like [yatm](https://github.com/audrow/yatm) parse the existing sitemap to generate issue tickets which results in duplicate tickets that point to the same URL thanks to the redirects. Removing duplicates from the sitemap would also help remove older pages from search engine lookups.

## How we generate the sitemap.xml
Presently the sitemap is generated using the [sphinx_sitemap](https://github.com/ros2/ros2_documentation/blob/67a35a42e5383cc1f7b5c3a373110b00a872fa62/conf.py#L84) extension for Sphinx. The implementation of this extension adds hooks to two events, `html-page-context` and `build-finished`, emitted by sphinx as seen [here](https://github.com/jdillard/sphinx-sitemap/blob/3bd87ef166bc9f3758d360ad4d23b524668b161d/sphinx_sitemap/__init__.py#L50-L51). The first hook simply adds the `pagename` to a queue while the second generates xml elements by popping `pagenames` from the queue after build is finished.

The conf.py script here defines a custom directive which also registers a hook for the event `html-collect-pages` as seen [here](https://github.com/ros2/ros2_documentation/blob/67a35a42e5383cc1f7b5c3a373110b00a872fa62/conf.py#L184-L185). The callback for this hook [returns an iterable](https://github.com/ros2/ros2_documentation/blob/67a35a42e5383cc1f7b5c3a373110b00a872fa62/conf.py#L249) that contains the new pages to generate for the redirect links. **The output of this generator will lead to the `html-page-context` event to trigger to build these new pages and since we also have the `sphinx_sitemap` extension, it will also trigger the hook that will add the redirect url to the list of pages to render into the sitemap.**

## Proposal to remove duplicates
With my limited understanding, I don't see a way to decouple the page generation and the sitemap generation unless we rely on our own sitemap extension. This extension would be very similar to `sphinx_sitemap` but instead of always appending `pagenames` into the queue, it can check for some `key : value` in the `context` arg of the callback to decide whether or not to append the pagename. The `RedirectFrom::generate()` impl will also be updated to append the required `key : value` [here](https://github.com/ros2/ros2_documentation/blob/67a35a42e5383cc1f7b5c3a373110b00a872fa62/conf.py#L237-L248).

Open to other suggestions!

clalancette, 2023-06-13:
Looking at the implementation of `sphinx_sitemap`, it does indeed look like there is no way to exclude certain pages (which I'm surprised about).  Also, looking at the implementation of `sphinx` it does look like there is an easy way to disable the emitting of the `html-page-context` event either.

So with that said, I agree with you that the best way forward here is to vendor our own copy of `sphinx_sitemap` with the changes we need.  The good news is that the [implementation](https://github.com/jdillard/sphinx-sitemap/blob/master/sphinx_sitemap/__init__.py) is a single file, so it shouldn't be too hard to vendor.  One request I'd make is to try to put this in a separate file, rather than in `conf.py`; that thing is already large.

Thanks for looking into this!

clalancette, 2023-06-22:
This was done in #3737, so closing.

closed 2023-06-22 by clalancette