ros2/ros2_documentation #3736: make -j `nproc` deb-pkg error

shuaixSu, 2023-06-14:


clalancette, 2023-06-14:
@YacobBY @razr Could you maybe take a look here?  I think you were the last two to touch that particular tutorial.

YacobBY, 2023-06-14:
@clalancette They keep doing slight updates to the deb file and deleting the old version link so they break every piece of code relying on it. 

I recommend just replacing the enitre page with "Install the Xanmod RT kernel" and deleting it. Else just replace the tutorial text with whatever link here https://cdn.kernel.org/pub/linux/kernel/projects/rt/5.4/ corresponds to the current version of the current kernel and wait for it to break again.

clalancette, 2023-06-14:
I guess we could also suggest to install the Ubuntu real-time kernel, though I think that requires a subscription.

Either way, would you mind submitting an update to that page which basically just removes the current content and adds these options instead?

razr, 2023-06-14:
I think Ubuntu's real-time kernel would be the best option to try