ros2/ros2_documentation #3741: ModuleNotFoundError: No module named 'urdf_tutorial_r2d2.state_publisher'

JHPHELAN, 2023-06-17:
Environment: Dell laptop / Ubuntu 22.04 / ROS2 Humble
[https://docs.ros.org/en/humble/Tutorials/Intermediate/URDF/Using-URDF-with-Robot-State-Publisher.html#view-the-results](url)
Section 7 View the results:
[ROS2 urdf_tutorial Experience 2023.06.17.pdf](https://github.com/ros2/ros2_documentation/files/11779851/ROS2.urdf_tutorial.Experience.2023.06.17.pdf)
`ubuntu@UbuntuBox:~/second_ros2_ws$ ros2 launch urdf_tutorial_r2d2 demo.launch.py`
```
[state_publisher-2] Traceback (most recent call last):
[state_publisher-2]   File "/home/ubuntu/second_ros2_ws/install/urdf_tutorial_r2d2/lib/urdf_tutorial_r2d2/state_publisher", line 33, in <module>
[state_publisher-2]     sys.exit(load_entry_point('urdf-tutorial-r2d2', 'console_scripts', 'state_publisher')())
[state_publisher-2]   File "/home/ubuntu/second_ros2_ws/install/urdf_tutorial_r2d2/lib/urdf_tutorial_r2d2/state_publisher", line 25, in importlib_load_entry_point
[state_publisher-2]     return next(matches).load()
[state_publisher-2]   File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
[state_publisher-2]     module = import_module(match.group('module'))
[state_publisher-2]   File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
[state_publisher-2]     return _bootstrap._gcd_import(name[level:], package, level)
[state_publisher-2]   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
[state_publisher-2]   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
[state_publisher-2]   File "<frozen importlib._bootstrap>", line 1004, in _find_and_load_unlocked
[state_publisher-2] ModuleNotFoundError: No module named 'urdf_tutorial_r2d2.state_publisher'
[ERROR] [state_publisher-2]: process has died [pid 27350, exit code 1, cmd '/home/ubuntu/second_ros2_ws/install/urdf_tutorial_r2d2/lib/urdf_tutorial_r2d2/state_publisher --ros-args -r __node:=state_publisher'].
[robot_state_publisher-1] [INFO] [1687029291.601515720] [robot_state_publisher]: got segment axis
...
```

clalancette, 2023-06-20:
So this looks like you probably forgot one of the steps during the installation process.  In particular, did you follow all of the instructions to edit the `setup.py` in https://docs.ros.org/en/humble/Tutorials/Intermediate/URDF/Using-URDF-with-Robot-State-Publisher.html#edit-the-setup-py-file ?

clalancette, 2023-10-03:
No response in several months, so closing.  If you are still having problems, please open a new question on https://robotics.stackexchange.com/questions/tagged/ros, which is our central Question and Answer site.

closed 2023-10-03 by clalancette