ros2/ros2_documentation #3757: Rosdep tutorial does not mention how to install rosdep

Ryanf55, 2023-06-23:
In the [rosdep tutorial](http://docs.ros.org/en/iron/Tutorials/Intermediate/Rosdep.html), it doesn't say how to install rosdep. 

Yea, I can install ros-dev-tools, if I'm on a supported distro, but I'm not. 

In the [rosdep readme](https://github.com/ros-infrastructure/rosdep/tree/master), it links to a very out of date [docs](http://docs.ros.org/en/independent/api/rosdep/html/) which recommend installing rosdep with `sudo pip ...`.

I'm not sure where the right place to update this; I'm thinking the tutorial should at least include instructions on where to get rosdep.

christophebedard, 2023-06-23:
> Yea, I can install ros-dev-tools, if I'm on a Ubuntu 23 right now, I can't do that (yet).

I think the expectation in these tutorials is that you're on a supported Ubuntu distro.

Ryanf55, 2023-06-23:
I'll file an issue on `rosdep`. 

From the binary installation tutorial, there is one word to change. I'll issue a PR.

closed 2023-06-23 by Ryanf55