ros2/ros2_documentation #3761: incorrect command in tutorial

cccccmake, 2023-06-27:
In the tutorial for ROS2 humble, section Gazebo, the command for visualizing lidar data in riv2 is:

`ros2 run ros_gz_bridge parameter_bridge /lidar2@sensor_msgs/msg/LaserScan[ignition.msgs.LaserScan --ros-args -r /lidar2:=/laser_scan`

This command is considered as a "bad pattern". However, the following command works:

`ros2 run ros_gz_bridge parameter_bridge /lidar2@sensor_msgs/msg/LaserScan@ignition.msgs.LaserScan --ros-args -r /lidar2:=/laser_scan`

ahcorde, 2023-06-27:
can you open a PR fixing this issue?

edgar-mycrobez, 2025-01-08:
Hello, I also followed the instruction given [here](https://docs.ros.org/en/humble/Tutorials/Advanced/Simulators/Gazebo/Gazebo.html) and I do get the "bad pattern" error when using the unidirectional command [ . For me it also makes sense to implement a unidirectional communication channel for this example, but it only works for me with the @ option. I am using Ubuntu 22.04, ROS Humble and Gazebo Fortress. Recommendation for using Gazebo Fortress from [here](https://gazebosim.org/docs/latest/getstarted/). Thank you.

ahcorde, 2025-01-14:
I tried this command in docker container with humble and Fortress. I don't have any issues.

Can you provide more details ?

edgar-mycrobez, 2025-02-09:
Hello, actually I would not  know what more details to add. I have tried this on a Dell XPS 13 with Ubuntu 22.04 LTS. I installed ROS2 according to [this](https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debs.html).

hx652, 2025-03-13:
If you still have the problem “bad pattern”, that's might because zsh treat "[" as something like regular expression, use`\[` may solve this. Sorry for bad English

edgar-mycrobez, 2025-03-13:
Hi hx652, all good with your English. It was perfectly understandable and this solves the problem. The detail on my side I did not mention was, that I am using zsh instead of bash. In **bash** the command

`ros2 run ros_gz_bridge parameter_bridge /lidar2@sensor_msgs/msg/LaserScan[ignition.msgs.LaserScan --ros-args -r /lidar2:=/laser_scan`

works perfectly well and in **zsh** I have to use the escape char `\` in order for it to work

In bash it works in both ways, using `\[` or `[` only

I have tested also the connector `]` in both, bash and zsh. It works without the escaoe char`\` on both.

For zsh users having to use the escape char `\` for this `[` and not for this `]` is not good. It is inconsistent and if trying to use command line examples that work in bash, you probably get the bad pattern error very often.

Thanks.