ros2/ros2_documentation #3762: Create rmw implementation tutorial

christophebedard, 2023-06-27:
The goal of this tutorial is to explain the rmw interface and describe how to implement it to support a middleware. To help illustrate this, [rmw_email](https://github.com/christophebedard/rmw_email) will be used as a "toy" example.

This issue is a WIP. An outline will be created for the tutorial and then some feedback will be requested before actually starting to write the full tutorial.

gavanderhoorn, 2024-05-15:
@christophebedard: have you had any time to work on this?

christophebedard, 2024-05-15:
Unfortunately no. I had to put it aside last year and have been pretty busy this year too. I hope to work on it over the next few months!

christophebedard, 2025-04-05:
> I hope to work on it over the next few months!

Famous last words.

closed 2025-10-16 by christophebedard