ros2/ros2_documentation #3770: Update documentation for Debian

Lecrapouille, 2023-07-02:
Hi, 

I have Debian 11, but I cannot follow your installation steps https://docs.ros.org/en/iron/Installation/Ubuntu-Install-Debians.html (other ROS2 version have the same issue) since steps are mainly Ubuntu based (for example Universe is not related to Debian). As final, I cannot install `sudo apt install ros-iron-desktop` since I have only `ros-desktop` which is for ROS1.

clalancette, 2023-07-02:
We don't provide binary installation for Debian.  In order to work on Debian, you'll have to follow the instructions on [building from source](http://docs.ros.org/en/iron/Installation/Alternatives/Ubuntu-Development-Setup.html).  There may be slight differences from Ubuntu as well.

Lecrapouille, 2023-07-02:
@clalancette Sad! The title `Ubuntu (Debian)` is confusing because we can suppose they both work. Could also be nice to add your sentence in the page.

clalancette, 2023-07-02:
> @clalancette Sad! The title `Ubuntu (Debian)` is confusing because we can suppose they both work. Could also be nice to add your sentence in the page.

We wanted to keep the title short in the sidebar.  It really should be `Ubuntu (Debian packages)`, which is maybe a better way to describe it.  If you'd like to make that update, and also update it to `RHEL (RPM packages)` at the same time, I'd be happy to review it.

Lecrapouille, 2023-07-03:
@clalancette I made a PR for the doc. Concerning the compilation, Debian I've almost done (Qt5 has usual, takes problem ^^) so compiling on Debian seems not taking so much issues, therefore creating deb files could be made. I did not find where is the github actions/CI/OpenBuild scripts creating Ubuntu packages. Do you know where they ztr, I may help to create script for making debian packages.

clalancette, 2023-07-04:
> I did not find where is the github actions/CI/OpenBuild scripts creating Ubuntu packages. Do you know where they ztr, I may help to create script for making debian packages.

We use a [Jenkins server](https://build.ros2.org) to create the packages we use for Ubuntu.  There is currently some thought about adding in Debian to the mix of packages we build, though it is not something we do lightly.  It adds a lot of additional overhead to the buildfarm, to the core, and to all package maintainers.  Pinging @nuclearsandwich and @tfoote who have been the ones looking into adding Debian as a target.

Lecrapouille, 2023-07-04:
Ok thanks, found repo and ticket https://github.com/ros2/ci/pull/355

closed 2023-07-04 by Lecrapouille