ros2/ros2_documentation #3785: eval incorrectly listed as unchanged from ROS 1

mjforan, 2023-07-13:
The [launch migration guide](http://docs.ros.org/en/rolling/How-To-Guides/Launch-files-migration-guide.html?highlight=env#replacing-an-include-tag) incorrectly states "eval and dirname substitutions haven’t changed." Eval statements do often need some syntax changes while migrating. See [this issue](https://github.com/ros2/launch/issues/469#issuecomment-720579449) for an example of the confusion this causes.

clalancette, 2023-07-20:
Yes, that's true, there are some differences here (particularly in the escaping).  If you'd like to contribute a pull request to improve that, it would be much appreciated.

mjforan, 2023-07-24:
I was hoping someone knowledgeable could give a more comprehensive list of changes. I created a pull request which adds a note about escape characters.

PR #3793 (2023-08-04): clarify eval escaping
Resolves #3785 - I don't know what other differences there may be, but at least this addresses one source of confusion.
closed 2023-08-04 by clalancette