ros2/ros2_documentation #3822: Where to mention community working groups?

ct2034, 2023-08-07:
> * Removes community working groups. They will find a new landing place shortly.
(https://github.com/ros2/ros2_documentation/pull/3374#:~:text=community%20working%20groups.-,They%20will%20find%20a%20new%20landing%20place%20shortly.,-Per%20Feb23%20TSC)

I am looking for a place to mention the [deliberation working group](https://docs.google.com/document/u/1/d/13TCHp3ycbDa2agGs0SjP3CpTllUeguAcXqih5LlCVDg), which is a community working group. Is there a new place for them, that I am just not able to find? :slightly_smiling_face:

clalancette, 2023-08-07:
These used to live on http://docs.ros.org/en/rolling/The-ROS2-Project/Governance.html , but they were removed with a new place in mind.  I don't know what that place is, though.  @kscottz @gbiggs can you please comment?

kscottz, 2023-08-07:
I believe we removed the community working groups but I am not sure if we actually created a new place for them.

PR #3884 (2023-08-30): Cleanup Working Groups
This pull request:

* Closes #3822 
* Adds a working group page.
* Links that page into the docs
* Clarifies community working groups.
* Adds existing working group with as much info as I could find. 

We'll need to ping the wg leaders to update the information.
closed 2023-08-30 by clalancette