ros2/ros2_documentation #3838: Unable to locate package ros-humble-desktop and ros-humble-ros-base

r8drfn89, 2023-08-17:
I typed the command sudo apt install ros-humble-desktop and get a unable to locate package error. Same thing happens if I do sudo apt install ros-humble-ros-base. 
[image]

KozhaAkhmet, 2023-08-18:
Which Ubuntu are you using? Maybe It is not compatible with your os.

r8drfn89, 2023-08-18:
@gavanderhoorn I have posted this on robotics.stackexchange.com. No body has responded to it.

r8drfn89, 2023-08-19:
It was my platform that was causing the error. I could not install the ros2 with the platform I had armhf. So I ended up re flashing the sd card with ubuntu 20.04.

closed 2023-08-19 by r8drfn89