ros2/ros2_documentation #3857: Makefile:19: update target 'html' due to: target does not exist python3 -m sphinx -M html "source" "build" -c .

songyuc, 2023-08-22:
Hi there!
I'm trying to contribute to the documentation and encountered an error when running `make -d html`:
```bash
PS D:\Professional\ROS2\ROS_documentation\ros2_documentation> make -d html
Finished prerequisites of target file 'html'.
Must remake target 'html'.
Makefile:19: update target 'html' due to: target does not exist
python3 -m sphinx -M html "source" "build" -c .
CreateProcess(NULL,python3 -m sphinx -M html source build -c .,...)
Putting child 01d9bfb0 (html) PID 31074688 on the chain.
Live child 01d9bfb0 (html) PID 31074688
Main thread handle = 000002a8
Reaping losing child 01d9bfb0 PID 31074688
make: *** [Makefile:19: html] Error 9009
Removing child 01d9bfb0 PID 31074688 from chain.
```
So how can I solve this?

Can anyone help me resolve this issue?
Any guidance would be greatly appreciated!

clalancette, 2023-08-22:
I've never built the documentation on Windows, so I'm not sure.  One guess is that on Windows, the python interpreter is usually `python`, not `python3`.  So you could try to edit the Makefile and change the `python3 -m sphinx` to `python -m sphinx` and see if that helps.

songyuc, 2023-08-22:
Hi, @clalancette!
Does ros2_documentation support Win10 or not?

clalancette, 2023-08-22:
> Does ros2_documentation support Win10 or not?

We don't usually build with it, so I would say not.  That said, if you find ways to make it work we'd be happy to review fixes for Windows.

songyuc, 2023-08-23:
OK, I will try to do it.

cychitivav, 2023-08-28:
> > Does ros2_documentation support Win10 or not?
> 
> We don't usually build with it, so I would say not. That said, if you find ways to make it work we'd be happy to review fixes for Windows.

No, this repository doesn't have support for Windows. In order to make it work on Windows, you need to configure the `make.bat` file (which used to come by default with Sphinx but was removed at some point). If you want to see the structure of the `make.bat` file, you can run `sphinx-quickstart` (included by default in the Sphinx installation) and configure it with the extensions added in this repository. However, to be honest, that can be quite complicated, and I would recommend using Docker with WSL (Windows Subsystem for Linux).

songyuc, 2023-08-29:
Thank you for your suggestion.
I think [clalancette](https://github.com/clalancette)'s solution is right, in which replacing **python3** with **python** in the Makefile can make it work.

songyuc, 2023-11-09:
Hi, guys,
Would you consider some Makefile tricks to make it work on Windows?

clalancette, 2023-11-09:
> Would you consider some Makefile tricks to make it work on Windows?

Yep, that would be fine.  As long as it continues to work on Linux (which is primary development/deployment platform), then anything additional to make it work on Windows would be appreciated.

songyuc, 2023-11-10:
Hello, guys,

Regarding this issue, I'd like to propose two potential solutions for your consideration:
1.  **Makefile Compatibility Improvement**: Modify the existing Makefile to include a system detection logic. Depending on the operating system (Windows or Linux/Unix), it could execute either `python -m sphinx` or `python3 -m sphinx`.
2.  **Separate Makefile for Windows**: Create a new Makefile specifically for Windows environments. This dedicated Makefile would use commands and configurations suitable for Windows, such as using `python -m sphinx`.

Both approaches have their advantages and challenges. I would appreciate your insights on which solution would align better with the project's maintenance strategy and overall workflow. Your guidance will help in making an informed decision that contributes positively to the project.

Thank you for considering these suggestions.
Best regards,
Yucheng Song

clalancette, 2023-11-10:
> 1. **Makefile Compatibility Improvement**: Modify the existing Makefile to include a system detection logic. Depending on the operating system (Windows or Linux/Unix), it could execute either `python -m sphinx` or `python3 -m sphinx`.

I think I would prefer this, *if* we can make it happen.  If we can't, then I'm fine with a separate Makefile.windows or whatever.

songyuc, 2023-11-13:
Thanks for your guidance.
And here is a simple code snippet for realizing this:
```
UNAME := $(shell uname)

PYTHON := python3
ifeq ($(UNAME), Windows)
    PYTHON := python
endif

BUILD      = $(PYTHON) -m sphinx
```

songyuc, 2023-11-13:
Hi, guys,
I have tested this trick on my Windows machine, which works fine.
Would you consider review this kind of PR?

Your guidance will be appreciated!

clalancette, 2023-11-13:
> I have tested this trick on my Windows machine, which works fine.

That seems reasonable to me.  Please go ahead and open a PR for that.

songyuc, 2023-11-14:
Hi @clalancette,

I have just submitted a PR to implement the Makefile modification we discussed:
https://github.com/ros2/ros2_documentation/pull/4013

Could you please take a look and review this PR?
Your feedback would be greatly appreciated to help get this Windows support change merged.
Let me know if you need any other information from me.

closed 2023-11-18 by songyuc