ros2/ros2_documentation #3859: Consolidate migration guides

mjforan, 2023-08-22:
There are a few migration guides under "How-To Guides" and a few under "The ROS 2 Project / Contributing". I propose creating a "ROS 1 Migration" folder under "How-To Guides" and grouping all the relevant content there. The migration guides have broader applicability than just people directly contributing to ROS 2 so they should be reorganized to a more visible path.

Does that sound reasonable?

clalancette, 2023-08-22:
Yes, completely agreed.  We even started on a PR to do exactly that: https://github.com/ros2/ros2_documentation/pull/3299 .  But we ran out of time to push that forward.  I'm fine with taking the existing documentation and moving it into a folder for now, and then adding content over time.

clalancette, 2023-08-31:
We ended up doing this in #3861, so closing this out.

closed 2023-08-31 by clalancette