ros2/ros2_documentation #3864: rviz plugin tutorial

LeweC, 2023-08-24:
It is currently very hard to find working examples or tutorials for custom rviz plugins.
As mentioned in #3242 and #3384 

At some point I got lucky and found a very good [repo](https://github.com/ros-visualization/visualization_tutorials/tree/ros2) for this, with many different kinds of rviz plugins. Also in many different ROS versions. But it was not easy to find.

I think it would make sense to include it here so that other people can find it more easily.

I would be happy to create a pull request if someone can give me some pointers where and how to include them here.

clalancette, 2023-08-31:
I think it would be great to add in those examples somewhere.  We just want to make sure that they are clearly marked as community-contributed examples, but otherwise please feel free to open a PR adding them.  Thanks!

DLu, 2024-07-30:
See also https://github.com/ros2/ros2_documentation/pull/3384

jkaniuka, 2025-01-14:
@clalancette I think we can close this one, because now we have tutorials both for RViz Panel (https://github.com/ros2/ros2_documentation/pull/4869) and RViz Display (https://github.com/ros2/ros2_documentation/pull/3384)

closed 2025-01-14 by ahcorde