ros2/ros2_documentation #3868: [Win10/Iron] Cannot get message of "[INFO] [custom_minimal_param_node]: Hello earth!" after using the launch file

songyuc, 2023-08-28:
Hi, guys!
I am learning about [《Using parameters in a class (Python) — ROS 2 Documentation: Iron documentation》](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Using-Parameters-In-A-Class-Python.html).

For the section of [[3.2 Change via a launch file]](https://docs.ros.org/en/iron/Tutorials/Beginner-Client-Libraries/Using-Parameters-In-A-Class-Python.html#change-via-a-launch-file), after running the
```bash
ros2 launch python_parameters python_parameters_launch.py
```
The CMD console did not print the "[INFO] [custom_minimal_param_node]: Hello earth!", as
```bash
D:\ros2_ws>ros2 launch python_parameters python_parameters_launch.py
[INFO] [launch]: All log files can be found below C:\Users\songyuc\.ros\log\2023-08-28-14-26-13-306213-DESKTOP-OB91EPG-21596
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [minimal_param_node.EXE-1]: process started with pid [6008]
```
, and it just stuck there with no action.

Could please help me solve this problem?

clalancette, 2023-08-28:
Please open questions like this on https://robotics.stackexchange.com/questions/tagged/ros, which is our central Question and Answer site.  You'll get a better answer there, and it will be searchable for the future.

Make sure to include a lot of information on what platform you are using, which ROS distribution you are using, and the exact steps you took.

closed 2023-08-28 by clalancette

songyuc, 2023-08-29:
Hi, @clalancette, thank you for your suggestion. I have asked my question there as https://robotics.stackexchange.com/q/103778/36083.