ros2/ros2_documentation #3893: Windows (binary) ROS 2 dependency installation Chocolatey Error

owuraku-zenas, 2023-08-30:
Windows (binary) — ROS 2 Documentation: Iron  documentation has a step for installing dependencies
https://docs.ros.org/en/iron/Installation/Windows-Install-Binary.html#install-dependencies

This step has a chocolatey command for installing some extra dependencies which should be downloaded from https://github.com/ros2/choco-packages/releases/tag/2022-03-15.
```choco install -y -s <PATH\TO\DOWNLOADS\> asio cunit eigen tinyxml-usestl tinyxml2 bullet```

The above code didn't work and I found this to work for me. The list of packages should come before the command for specifying the location.
```choco install -y asio cunit eigen tinyxml-usest1 tinyxml2 bullet -s <PATH\TO\DOWNLOADS\>```

This must be updated in the docs.

clalancette, 2023-08-30:
The command as-is seems to work fine for me with chocolatey version 2.2.0.  What version are you using?  Also, what was the error message you got?

owuraku-zenas, 2023-08-31:
Sorry about the confusion. I just tried it again and it worked. I'm using version 2.2.2. Thank you.

closed 2023-08-31 by owuraku-zenas